Legion2/CorsairLightingProtocol #170: Error compiling for board CLP SparkFun Pro Micro 5V, 16 MHz

davidelolli, 2020-08-13:
Good morning developers, I really love your idea and I'm trying to install it on my Pro Micro board,
I'm not an expert in Arduino and I've followed the guide and I've tried different Arduino versions on windows and Mac OS, but I receive always this problem, can someone help me? 

Thanks for your work

`Arduino: 1.8.0 (Mac OS X), Board: "CLP SparkFun Pro Micro 5V, 16 MHz, Lighting Node PRO"

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0xa9f2a]

goroutine 1 [running]:
panic(0x153640, 0xc42000e050)
	/usr/local/go/src/runtime/panic.go:500 +0x1a1
arduino.cc/builder.(*SetupBuildProperties).Run(0x247c18, 0xc42006d180, 0x247c18, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/setup_build_properties.go:83 +0xf8a
arduino.cc/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x247c18, 0xc42006d180, 0x42055555, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/container_setup.go:59 +0x3c3
arduino.cc/builder.runCommands(0xc42006d180, 0xc42003bcf8, 0x3, 0x3, 0xc42000ce01, 0x1, 0x21a960)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:191 +0xcd
arduino.cc/builder.(*ParseHardwareAndDumpBuildProperties).Run(0xc42003bd58, 0xc42006d180, 0xc42003bda8, 0x15ea80)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:180 +0xfb
arduino.cc/builder.RunParseHardwareAndDumpBuildProperties(0xc42006d180, 0xc42003bda8, 0x247c18)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:227 +0x35
main.main()
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/arduino-builder/main.go:307 +0xcaa
/private/var/folders/m2/3x6j0g_x6xb515sl94fky2yr0000gn/T/AppTranslocation/277FA101-4204-4386-B742-868312699271/d/Arduino.app/Contents/Java/arduino-builder returned 2

Error compiling for board CLP SparkFun Pro Micro 5V, 16 MHz.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

Legion2, 2020-08-13:
Could you try and update your Arduino IDE version?

davidelolli, 2020-08-16:
> Could you try and update your Arduino IDE version?

Sure, the problem started with the latest version of Arduino, i've also tried to downgrade in different machines but no luck

Legion2, 2020-08-16:
From your log (`Arduino: 1.8.0 (Mac OS X)`) I would tell you are not using the latest version, which is Arduino 1.8.13. So, please use the latest version and if it still gives you the error, enable "Show verbose output during compilation" option in File -> Preferences.

davidelolli, 2020-08-17:
Thanks again for your support, I've update and enabled the option in Arduino, but same error:

`Arduino: 1.8.13 (Mac OS X), Board: "CLP SparkFun Pro Micro 5V, 16 MHz, Lighting Node PRO"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/davidelolli/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/davidelolli/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/davidelolli/Documents/Arduino/libraries -fqbn=Legion2:avr:promicro5vclp:device=lnp -ide-version=10813 -build-path /var/folders/m2/3x6j0g_x6xb515sl94fky2yr0000gn/T/arduino_build_690051 -warnings=none -build-cache /var/folders/m2/3x6j0g_x6xb515sl94fky2yr0000gn/T/arduino_cache_184918 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/davidelolli/Documents/Arduino/libraries/Corsair_Lighting_Protocol/examples/LightingNodePRO/LightingNodePRO.ino
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x50 pc=0x16be5f3]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.(*SetupBuildProperties).Run(0x1eead88, 0xc000055400, 0x1eead88, 0xc000210270)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/setup_build_properties.go:89 +0x10a3
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x1eead88, 0xc000055400, 0x0, 0x0)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/container_setup.go:87 +0x464
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc000055400, 0xc000141cc0, 0x2, 0x2, 0x0, 0x0)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder.go:192 +0x19f
github.com/arduino/arduino-cli/legacy/builder.(*ParseHardwareAndDumpBuildProperties).Run(0xc000145d40, 0xc000055400, 0x1eeac70, 0xc000271550)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder.go:183 +0x96
github.com/arduino/arduino-cli/legacy/builder.RunParseHardwareAndDumpBuildProperties(...)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder.go:215
main.main()
	/Users/jenkins/Jenkins/workspace/arduino-builder-mac-catalina/main.go:390 +0x879
/Applications/Arduino.app/Contents/Java/arduino-builder returned 2

Error compiling for board CLP SparkFun Pro Micro 5V, 16 MHz.
`

Legion2, 2020-08-17:
You are using a SparkFun Pro Micro, did you install https://github.com/sparkfun/Arduino_Boards#installation-instructions ?

PS: use ``` for multi line code blocks

davidelolli, 2020-08-17:
yes, iv'e tried another time, but nothing

Legion2, 2020-08-17:
Have you tried to upload some other simple sketch to the Arduino, e.g. the blink example sketch?

jath03, 2020-08-19:
I'm getting the same issue with Arduino 1.8.13 on Arch Linux.  I don't think it's an uploading problem, because it happens even when there is no board plugged in.  If this makes a difference, I'm not actually using a Pro Micro.  I'm using adafruit's itsybitsy 5v 16 MHz, which seems like the same thing, with a few different pins.  I was hoping it would just work, but haven't tested it yet.  I'm pretty sure this issue is unrelated to the board mismatch, becasue it doesn't matter if there is a board plugged in or not.  
```
Arduino: 1.8.13 (Linux), Board: "CLP SparkFun Pro Micro 5V, 16 MHz, Lighting Node PRO"

/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/jack/.arduino15/packages -tools /usr/share/arduino/tools-builder -tools /home/jack/.arduino15/packages -libraries /home/jack/Arduino/libraries -fqbn=Legion2:avr:promicro5vclp:device=lnp -ide-version=10813 -build-path /tmp/arduino_build_12536 -warnings=none -build-cache /tmp/arduino_cache_95655 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/jack/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/jack/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/home/jack/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/jack/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/home/jack/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/jack/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/jack/Arduino/LightingNodePRO/LightingNodePRO.ino
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x50 pc=0x55d7dbb9576a]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.(*SetupBuildProperties).Run(0x55d7dc3bdca8, 0xc00019a800, 0x55d7dc3bdca8, 0xc000432030)
	github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/setup_build_properties.go:89 +0x10ca
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0x55d7dc3bdca8, 0xc00019a800, 0x0, 0x0)
	github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/container_setup.go:87 +0x45d
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc00019a800, 0xc0005b1db8, 0x2, 0x2, 0x0, 0x0)
	github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder.go:192 +0x176
github.com/arduino/arduino-cli/legacy/builder.(*ParseHardwareAndDumpBuildProperties).Run(0xc0001e9e3e, 0xc00019a800, 0x55d7dc3bdb68, 0xc0003114c0)
	github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder.go:183 +0x98
github.com/arduino/arduino-cli/legacy/builder.RunParseHardwareAndDumpBuildProperties(...)
	github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder.go:215
main.main()
	github.com/arduino/arduino-builder/main.go:390 +0x914
/usr/share/arduino/arduino-builder returned 2

Error compiling for board CLP SparkFun Pro Micro 5V, 16 MHz.

```

Legion2, 2020-08-19:
@jath03 did you try to compile other code for that board? Just to make sure the error is really related to the code of this project.

jath03, 2020-08-19:
Hmm, this issue might belong in CorsairLightingProtocolBoards.  When I set the board to the correct value, the same program (the LightingNodePRO example) compiles fine.

jath03, 2020-08-19:
And the built-in BareMinimum example doesn't compile on the CLP SparkFun Pro Micro 5V, 16 MHz. Compiling the BareMinimum example fails with the same error on all the CLP boards except the CLP Arduino Leonardo, which compiles fine.

Legion2, 2020-08-20:
> When I set the board to the correct value, the same program (the LightingNodePRO example) compiles fine.

What do you mean with "correct value"?

> And the built-in BareMinimum example doesn't compile on the CLP SparkFun Pro Micro 5V, 16 MHz

Can you try to compile it for "SparkFun Pro Micro 5V, 16 MHz" without the CLP in the name.

jath03, 2020-08-20:
> What do you mean with "correct value"?

The official board from adafruit's boards package.

After getting the boards list from [here](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide/installing-mac--linux), The BareMinimum example compiled fine for the SparkFun Pro Micro, configured with the 5v 16 MHz processor

Legion2, 2020-08-20:
> The official board from adafruit's boards package.

Can you send me the link to it, is it different from the SparkFun Pro Mirco?

jath03, 2020-08-20:
It seems like adafruit's version of the pro micro.  It has the same processor, but a few different pins.  As much as I'd love to get this device supported, like I said earlier, I don't really think that is the problem with the compilation error.  
https://learn.adafruit.com/introducting-itsy-bitsy-32u4/arduino-ide-setup

Legion2, 2020-08-20:
> After getting the boards list from here, The BareMinimum example compiled fine for the SparkFun Pro Micro, configured with the 5v 16 MHz processor

Did you try to compile with "CLP SparkFun Pro Micro 5V, 16 MHz" after you have installed "SparkFun Pro Micro 5V, 16 MHz"?

jath03, 2020-08-20:
No, I didn't.  That fixed it!  You need to have the official board installed for the CLP one to work!

Legion2, 2020-08-20:
Yes this is also documented in the CorsairLightingProtocolBoards repo, but I thinks it's not placed very visible.

Legion2, 2020-08-20:
@jath03 if you want support for the Adafruit board please open an issue in the CorsairLightingProtocolBoards repo

leonardcoutinho, 2020-08-30:
try to compile for CLP Leonardo board, works for me, I have an Pro Micro 5v board too

sesheron, 2020-10-09:
> try to compile for CLP Leonardo board, works for me, I have an Pro Micro 5v board too

I had this problem, some of the clone boards are labeled as Pro Micro, but when you read their minimal instructions, they say to use Leonardo.

commit e8441242 (2020-11-13): added SparkFun Boards installation hint to readme

close #170
closed 2020-11-13 by Legion2