Legion2/CorsairLightingProtocol #171: Need more deatail about "Persistent settings for use without a USB connection."

LennyBo, 2020-08-14:
When I turn off my PC, 5V is still supplied to my USB ports. This makes it that I always have my LEDS on. I can disable this feature by enabling deep sleep but that would also disable my wake on lan setup I have and need. I saw that in the features section of the README there is mention of "Persistent settings for use without a USB connection." which is what I assume the problem in my case but haven't found any further explanation in the documentation or the readme.

Legion2, 2020-08-14:
> Persistent settings for use without a USB connection.

means that animations can be stored in the EEPROM of an arduino and are played back when powered without a connection to iCUE.

The solution to your question depends on how you power the LEDs, from what I understand from your description is, that you power them via the Arduino or an external power supply. If you power them via arduino I would suggest powering them from your PC power supply. Then the LEDs go off when the PC is off because the SATA and molex power should be off when the PC is off.

If you use an external power supply you should use it to turn off the LEDs. Alternatively you can set in iCUE the hardware lighting to a static black color.
[image]

LennyBo, 2020-08-14:
Hardware lighting did the trick Thanks
/close

closed 2020-08-14 by LennyBo