Legion2/CorsairLightingProtocol #175: Not smooth color effects transition with fixIcueBrightness enabled

leonardcoutinho, 2020-08-30:
I have an bug with fixIcueBrightness enabled, with color effects transition, like rainbow wave or color wave, I can see leds very bright but not smooth. When I set brightness in Icue to 25%, all effects with transition are smooth. And I don't notice bright difference between 25% and 100%. With fixIcueBrightness disabled, I can set brightness, and effects are smooth in all brightness level. I use ws2812b strips with 60leds/m, 90(channel 2)+(channel 2)80leds, with Arduino Leonardo Pro Micro board(5v). 

**To Reproduce**
Steps to reproduce the behavior:
1. Flash build with fixIcueBrightness enabled
2. Set effects to rainbow wave or color wave
3. Observe if effects are smooth
4. change brightness to 25% and look effects again.
5. Flash build with fixIcueBrightness disabled and check effects and brightness setting in Icue.

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Leonardo/Pro Micro
 - IDE: Arduino Ide
 - Version: 0.14.1
 - Sketch: LightingNodePRO

**Code changes**
fixIcueBrightness code enable in both channels

Legion2, 2020-08-30:
Did you use scaling or how did you configure your channels in iCUE? How many leds are configured in the arduino code? Which version of iCUE do you use?

leonardcoutinho, 2020-08-31:
> Did you use scaling or how did you configure your channels in iCUE? How many leds are configured in the arduino code? Which version of iCUE do you use?

I use latest iCue version(3.32.80) . I try to scale with normal function but not work right for me, then I use segment scaling and work nice. In channel 1 I put 80 leds, in 5 sections of 16leds each(and in iCue I configure for 5leds strips), and in channel 2 I put 90 leds, in 6 sections, 5 of then with 16leds each and 1 with 10leds, and in iCue I configure to 6 led strips. This setup works nicely. I power all of this off my PC power supply 5v rail(is an 600w 80plus psu). 

I test before configuring led strip as LL fan, but I think configure it as led strip is better to manage and visualize effects. 
Strange is if fixIcueBrightness disabled, leds transitions is smooth, and when is enabled I clearly see leds colors changing color from one led to other(with 100% brightness in iCue, or 50%, with 25% is smooth again). I try change code to setting brightness exponencially, but is bad too. 

Does brightness in 100% introduce some latency? what I can do for test?

Legion2, 2020-08-31:
I added a FPS counter in #176. Look at the examples how it is used https://github.com/Legion2/CorsairLightingProtocol/pull/176/files. For me it prints 24/25 fps which is totally smooth.

leonardcoutinho, 2020-09-01:
> I added a FPS counter in #176. Look at the examples how it is used https://github.com/Legion2/CorsairLightingProtocol/pull/176/files. You need to download the dev version of the library to use the FPS counter.
> 
> For me it prints 24/25 fps which is totally smooth.

I try to compile with PrintFPS function like in example, but I can't make work. In serial console only print "FPS: " and when print an number is 0. I try to compile DebugSketch but can't work too. I notice too is there an lag in leds when I flash my code with PrintFPS function. I think that can be a problem with my Arduino who can't handle much instructions.

Here is my code without PrintFPS which run smooth:

```

#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2
#define DATA_PIN_CHANNEL_2 3

CRGB ledsChannel1[80];
CRGB ledsChannel2[90];

CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
	
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 80);
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 90);
	ledController.addLEDs(0, ledsChannel1, 50);
	ledController.addLEDs(1, ledsChannel2, 60);
    
  ledController.onUpdateHook(0, []() {
    CLP::SegmentScaling segments[5] = {{10, 16}, {10, 16}, {10, 16}, {10, 16}, {10, 16}};
    CLP::scaleSegments(&ledController, 0, segments, 5);
  });
  ledController.onUpdateHook(1, []() {
    CLP::SegmentScaling segments[6] = {{10, 16}, {10, 16}, {10, 16}, {10, 16}, {10, 16}, {10, 10}};
    CLP::scaleSegments(&ledController, 1, segments, 6);
  });
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
   
	}
  
}
```

Legion2, 2020-09-02:
@leonardcoutinho sorry, I didn't tested it after I changed the fps counter input form seconds to milliseconds. But now I fixed and tested it #178.

leonardcoutinho, 2020-09-02:
I tested again and FPS monitor works now. With my code and without fixIcueBrightness:

FPS: 24.8
FPS: 24.8
FPS: 24.8
FPS: 24.8
FPS: 25.0
FPS: 24.8
FPS: 24.8
FPS: 24.8

With fixIcueBrightness enabled: 

FPS: 24.8
FPS: 24.8
FPS: 24.8
FPS: 25.0
FPS: 24.8
FPS: 24.8
FPS: 25.0

I make an video to show what is happen, is a very subtle lag when led colors is moving from one led to other:  https://youtu.be/V5m2r79VMtA

Legion2, 2020-09-02:
Strange, does this happen only on one channel or both? What happens if you remove the scaling but keep the fixIcueBrightness?

leonardcoutinho, 2020-09-02:
I tested in both channels, individually, and problem only happens in channel who has fixicuebrightness(if both channel has, occur in both channels). I tested too without scaling, only with fixIcueBrightness, and happens too.

leonardcoutinho, 2020-09-02:
I saw gammaCorrection and tested it with and without fixIcueBrightness, and have some results: 
With gammaCorrection function have a little lag, but smooth. With fixIcueBrightness have lag. Then I test with solid fixed color and change brightness slider in iCue. With nothing or with gammaCorrection, brightness slider works as expected. With fixIcueBrighness the brightness slider in 25% has same bright as slider in 100%, and in 50% its fade, and in 0% its turnoff as expected. Is strange to have 100% brightness level in 25% too, this problem can be my led strip. I will try change multiplier in fixIcuebrightness function to see whats happen.

leonardcoutinho, 2020-09-02:
I did this to print rgb brightness value in fixIcueBrightness variables.  
```
                       Serial.println(leds[ledIndex].r, 1);
			leds[ledIndex].r = leds[ledIndex].r * 1;
			Serial.println(leds[ledIndex].g, 1);
			leds[ledIndex].g = leds[ledIndex].g * 1;
			Serial.println(leds[ledIndex].b, 1);
			leds[ledIndex].b = leds[ledIndex].b * 1;
```

In serial console i can see values between 0 and 255. Apparently it as already converted from 127 to 255 format, transition from one led to other is smooth too, even using this function.
Then I modify multiplier to 2x again and see serial console. I can see values between 0 and 255 again, but this time I have lag in leds transition. Then I change multiplier to 0.5 and see again, this time has valuens between 0 and 127 and transition is smooth. 
I test bright slider in iCue too, with 2x multiplier it has problem from last post, but in 0.5 or 1x works as expected.

Legion2, 2020-09-02:
which iCUE version do you use?

leonardcoutinho, 2020-09-02:
I use 3.32.80, the last one

Legion2, 2020-09-02:
Ok I can reproduce it now.

Legion2, 2020-09-02:
Ok after some testing I found that iCUE only sends 50% brightness for LS100 and LT100. Far all other devices iCUE sends 100% brightness. So the fixIcueBrightness must/should only be used with LS100 and LT100. Otherwise you get artifacts because double the brightness when it is already at 100% will produce strange effects.

commit 3e91580f (2020-09-02): added note to fixIcueBrightness close #175
fixIcueBrightness should only be used with LS100 and LT100
PR #179 (2020-09-02): added note to fixIcueBrightness
`fixIcueBrightness` should only be used with LS100 and LT100 close #175
closed 2020-09-02 by Legion2