Legion2/CorsairLightingProtocol #184: Ambient light (LS100) not regonized

Justdigit, 2020-09-10:
**Describe the problem**
When i flash Ambientbacklight sketch or LS100 its always regonized as Lightning node pro.
I have changed the Arduino Device ID but it wont change 

**System (please complete the following information):**
 - OS: WIN10
 - Board: Arduino Pro micro
 - IDE: [e.g. Arduino IDE]
 - Version: [e.g. 0.3.0]
 - Sketch: Ambientbacklight

cjgun, 2020-09-10:
I did the same and everything worked out for me.
But I filled in the correct one first bootLoader from Leonardo.

Board: Arduino Pro micro
Sketch: Ambientbacklight

Tools — Boadr - CLP Arduino Leonardo
Device — LS100 Smart Lighting Controller

everything works as it should

Legion2, 2020-09-10:
Did you select the correct device?
[image]

Justdigit, 2020-09-11:
Who wants to give me a hammer??

Works fine..... Just have to rescale the leds to my stripcount. Going to dive in this evening.

Thanks.

closed 2020-09-11 by Justdigit