Legion2/CorsairLightingProtocol #185: Only a single LED lights green.

KBhamm, 2020-09-11:
**Describe the problem**
A clear and concise description of your problem.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
Windows 10
Arduino Leonardo
Arduino IDE
1.8.13
Node Pro

**Code changes**
Changed leds to SK6812 and numbers to 72 (also ws2812b wont work)

My problem is that the node works normal in icue but if i want to controll the LEDs only a single green LED lights up and i cant change it or controll it. i use a evenbetter light strip wich uses sk6812 leds these should be the same as ws2812b tells the datasheet. my hoodloader flashed uno stoped working so i bought a new genuine leonardo like in the tutorial and soldered the Data pin to the back out/in of the power jack like i saw in many arduino project and it worked fine with the uno before.

`#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2
#define DATA_PIN_CHANNEL_2 3

CRGB ledsChannel1[72];
CRGB ledsChannel2[72];

CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 72);
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 72);
	ledController.addLEDs(0, ledsChannel1, 72);
	ledController.addLEDs(1, ledsChannel2, 72);
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}`

Legion2, 2020-09-11:
Check that the strip is correctly connected to the correct pin of the arduino and also make sure the GND of the arduino is connected with the GND of the led strip. How do you power the led strip?

KBhamm, 2020-09-13:
It is connected the same way as before. The plus and ground pin is connected via a Power Supply to the led strip. The ground its connected thou a cable to the arguinos power jack ground. The data pin is soldered from pin 2 to the back port of the power jack. from there the data goes to the strip. It worked there before with my Hoodloader uno for nearly a year now. But the hoodloader uno made problems with the newest icue update.

Legion2, 2020-09-13:
I don't understand "The data pin is soldered from pin 2 to the back port of the power jack"

KBhamm, 2020-09-15:
[image]
i Hope this helps. With the old hoodloader uno this worked.

KBhamm, 2020-09-15:
The power jack is used via connector for ground and data.

Legion2, 2020-09-15:
Can you directly connect the data pin without using the power jack

closed 2020-11-02 by stale[bot]