Legion2/CorsairLightingProtocol #186: ATTiny85 Support

LuckyMallari, 2020-09-13:
Do you have any plans to support **[ATTiny85](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2586-AVR-8-bit-Microcontroller-ATtiny25-ATtiny45-ATtiny85_Datasheet.pdf)**? I tried to setup a Leonardo as a composite device to have multiple channels. Got it working but the leo simply is not powerful enough esp with the limited memory. My goal is to have 20 independent channels. With your code, this is possible with 10 separate devices emulating Node pro (2 channels each). But an ATTiny85 approach is many times less expensive. 

So now, the route I'm planning to take is to have several ATTiny85 chips instead, with optionally another board as a USB host (as a generic USB hub if cheaper).

Legion2, 2020-09-26:
The ATTiny85 doesn't support usb hid/it has no native usb support.

Legion2, 2020-09-26:
But the CLP library is modular so you can split the usb part from the led controller.

LuckyMallari, 2020-09-27:
> The ATTiny85 doesn't support usb hid/it has no native usb support.

V-USB runs well on the Attiny chip

Legion2, 2020-09-27:
Is V-USB available as arduino library?

Legion2, 2020-09-27:
>  I tried to setup a Leonardo as a composite device to have multiple channels. Got it working but the leo simply is not powerful enough esp with the limited memory.

@LuckyMallari Do you mean you have got it working to report more than one device in iCUE on a Leonardo?

closed 2020-11-14 by stale[bot]