Legion2/CorsairLightingProtocol #188: I only can use Lightning node pro...

ZeddBlock, 2020-09-19:
Hey !

First at all, forgive my english.
I'm using an arduino Micro (32u4). I tried this with the Lightning Node Pro, and everything work fine in Icue. 
But, when I choose LT100 ou LS100, my Arduino is still recognize as Nightning Node Pro... 

How to do ? Anyone else had this issue ?

cjgun, 2020-09-19:
> Hey !
> 
> First at all, forgive my english.
> I'm using an arduino Micro (32u4). I tried this with the Lightning Node Pro, and everything work fine in Icue.
> But, when I choose LT100 ou LS100, my Arduino is still recognize as Nightning Node Pro...
> 
> How to do ? Anyone else had this issue ?

Choose LS100 instead of Node Pro when choosing a board model.
Tools- Devices - LS100
I have the same scarf and everything works fine.

ZeddBlock, 2020-09-19:
Thank you so much ! It works. I just choosed it on exmaple. 
Have a nice day ;-)

closed 2020-09-19 by Legion2