Legion2/CorsairLightingProtocol #189: Possible to emulate 2 or more Lighting Nodes with 1 Arduino?

capullo, 2020-09-22:
Hi,

would it be possible to let 1 Arduino emulate more than 1 Lightning Node in parallel, to have multiple channels?

From USB/HID side it should be possible, or?

Legion2, 2020-09-23:
Maybe, but I don't know HID on Arduino very well. So I can't create multiple raw HID endpoints.

t0mb3r, 2020-10-03:
This would be great, would appreciate it if someone could shine a light on how to do it

Legion2, 2020-10-03:
We should not expect so much from such a feature, because the 32u4 memory is at its limit with two long led channels, so you can not control 4 channels with many leds. Also this would make the setup process more complicated because there is not 1:1 relation between arduino and corsair device anymore. Everything that is currently a singleton in the code (most of the RawHID code) has to be rewritten in an object oriented way which then also requires more memory. And we currently don't even know if iCUE can handle two HID devices which are hosted on the same USB device, because the HID device have to share the USB control endpoint.

When I think of the use case for this, I only can imagine someone wants to safe the money or the space of a second arduino. But when considering, that you can get a 32u4 board for ~10€/10$, e.g. Pro Micro Boards, and the small size of these boards, e.g. Piksey Atto Board, then there is not really a use case for this feature.

I'm going to close this issue because I don't see it as an improvement of the CLP library. If someone make it work, I may add it to the library.

closed 2020-10-03 by Legion2