Legion2/CorsairLightingProtocol #194: Scale of LED´s

TheDeepSpacer, 2020-10-02:
**Describe the problem**
Hi, in my case i want to illuminate all 300 LED. But im not able to to that.
Tried also use less then 300 but that also wont work.
Just only the few from the "standart" Scetch.

**System:**
 - OS: Windows 10 2004
 - Board: Pro Micro Modul ATmega32U4 5V, 16MHz
 - LEDS: 5m WS2812B -> 300 LED´s
 - IDE: Arduino IDE
 - Version: 1.8.13
 - Sketch: LightingNodePRO

Legion2, 2020-10-02:
Have you looked at this [example sketch](https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/RepeatAndScale/RepeatAndScale.ino#L39) which shows how to scale the led channel?

TheDeepSpacer, 2020-10-03:
My fault, used
`  ledController.onUpdateHook(1, []() {
    CLP::scale(&ledController, 0, 300);`
instead of
`  ledController.onUpdateHook(0, []() {
    CLP::scale(&ledController, 0, 300);`

closed 2020-10-03 by TheDeepSpacer