Legion2/CorsairLightingProtocol #197: Not able to use reverse function

adam120800, 2020-11-08:
Hi,

Unfortuanetly everything i try i am not able to use the "reverse" function to reverse the order of led channel.

I have tried:
 ledController.onUpdateHook(0, []() {
   CLP::reverse(&ledController, 0);

Legion2, 2020-11-08:
can you show me your complete code?

adam120800, 2020-11-08:
#include <CorsairLightingProtocol.h>
#include <FastLED.h>
#define BRIGHTNESS 115
#define DATA_PIN_CHANNEL_1 4
#define DATA_PIN_CHANNEL_2 2
#define DATA_PIN_CHANNEL_3 5
#define DATA_PIN_CHANNEL_4 3

CRGB ledsChannel1[60];
CRGB ledsChannel2[120];

CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
  FastLED.setBrightness(BRIGHTNESS);

  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 60);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel1, 60);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_3, GRB>(ledsChannel2, 120);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_4, GRB>(ledsChannel2, 120);
  ledController.addLEDs(0, ledsChannel1, 60);
  ledController.addLEDs(1, ledsChannel2, 60);  
  ledController.onUpdateHook(1, []() {
   CLP::scale(&ledController, 1, 120);
   CLP::reverse(&ledController, 1)
  });
}

void loop() {
  cHID.update();

  if (ledController.updateLEDs()) {
    FastLED.show();
  }
}

Legion2, 2020-11-08:
Reverse should be used before scale, else only the first 60 leds get reversed, after they have been scaled. If this is not your problem please describe what happens when you run this code and what do you expect. (please use the Support Issue template)

adam120800, 2020-11-08:
I have used reverse before scale and it works on Led channel 2 using this code:

Legion2, 2020-11-09:
Can you please describe what exactly you mean by "wont work", does it not light up, or is it not reverse, or something else? 

> I am using a wave ligthing effect for testing

Maybe use a static effect on the first leds, if it works the last leds should light up. Moving effects are difficult, because the direction can be changed in iCUE.

PS: use ``` to quote your code

adam120800, 2020-11-12:
Sure thing. When you have lightning node in corsair you cant change direction for Wave effect only on keyboards and other devices. It doesnt light up at all on Leds channel 1. Leds channel 1 has 60 leds set and doesnt even light up. When i use the reverse function on leds channel 2 which has 120 scalled it will properly show and reverse

if i use
ledController.onUpdateHook(0, {
CLP::reverse(&ledController, 0)
});

it wont work but if i use: 

ledController.onUpdateHook(1, {
CLP::reverse(&ledController, 1)
});

the effect will show and will be reversed just like i physically turned the strip 180 degrees.

Legion2, 2020-11-13:
I can not reproduce the error. Does the channel 1 light up if you don't use the reverse function? Please send me the whole arduino sketch which I can compile and upload to my arduino to debug the error. Your previous code snippets I could not compile because they missed some semicolons.

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: [e.g. Arduino Leonardo]
 - Device: [e.g. Lighting Node PRO]
 - Version of IDE: [e.g. Arduino IDE 1.8.13]
 - Version of CLP: [e.g. 0.14.2]
 - Version of CLP Boards: [e.g. 0.2.0]
 - Sketch: [e.g. RepeatAndScale]

closed 2020-12-31 by stale[bot]