Legion2/CorsairLightingProtocol #200: LT100 more leds

kdbhu, 2020-11-20:
Hi!
I tried use the library and evrthing was good.
I want to create my own LT100, but it only work with 10 leds (I try to write number of leds, bit nothing changed).
I use windows 10 with sparkfun pro micro.
Can u help me?

Legion2, 2020-11-20:
For iCUE the number of LEDs per tower is fixed, so you have to configure the number of LEDs for 2 towers. You can then scale the two towers using the `scaleSegments` function to whatever number you like, see https://github.com/Legion2/CorsairLightingProtocol/wiki/Scale-separate-segments-of-a-channel.

kdbhu, 2020-11-20:
Thanks for help. I will try it.

kdbhu, 2020-11-20:
Sorry i'm not very good at english can u send me a example code?

kdbhu, 2020-11-20:
> For iCUE the number of LEDs per tower is fixed, so you have to configure the number of LEDs for 2 towers. You can then scale the two towers using the `scaleSegments` function to whatever number you [...]

kdbhu, 2020-11-20:
So I have to define 2*27 leds?
Can u help in private?

Legion2, 2020-11-20:
the following code make the towers using 50 LEDs.
```c++
ledController.onUpdateHook(0, []() {
  CLP::SegmentScaling segments[2] = {{27, 50}, {27, 50}};
  CLP::scaleSegments(&ledController, 0, segments, 2);
});
```
Here is an example how the onUpdateHook should be used https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/RepeatAndScale/RepeatAndScale.ino

Also make sure `CRGB ledsChannel1[100];` has enough LEDs (50 + 50) or whatever number you need.

I can give more specific answers if you provide more information.

kdbhu, 2020-11-20:
Thank you! that's work very well.

Legion2, 2020-11-20:
change `ledController.addLEDs(0, ledsChannel1, 100)` to `ledController.addLEDs(0, ledsChannel1, 54)`

kdbhu, 2020-11-20:
So fixed it, but thanks for a lot of help!
I need zo change the value of ledControllers led.

`ledController.addLEDs(0, ledsChannel1, 54);`

closed 2020-11-20 by Legion2