Legion2/CorsairLightingProtocol #202: Sparkfun Pro Micro Driver error

Wiedenhoferkilian, 2020-12-03:
Hello,
My Problem is that i wanted to do this project with a arduino nano. But the nano is not supported so i ordered a Sparkfun pro Micro with 3.3V and 8mhz. When i first plugged it in windows said that there is a driver issue but in the arduino ide it apperied so i uploaded the sketch for the node pro but now the pro micro is not visiable in the arduino ide and so not visable in icue. I already have tried to install the drivers from sparkfun but windows always says that the best drivers are already installed. Ofcourse i already have removed the drivers and reinstalled them. Can pls someone help me?

Legion2, 2020-12-03:
Try the first item of the troubleshooting https://github.com/Legion2/CorsairLightingProtocol/wiki/Troubleshooting#i-cant-upload-a-new-sketch-to-my-sparkfun-pro-micro

Legion2, 2020-12-11:
Did this resolve the problem for you?

closed 2020-12-23 by Legion2