Legion2/CorsairLightingProtocol #205: LT100

kdbhu, 2020-12-25:
Hi!
I try create a LT100, but ICUE only detect 1 tower and the second strip show a random light effect. 
Can somebody help me to ffix this?

Here is my code:
```
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 2
#define DATA_PIN_CHANNEL_2 3

CRGB ledsChannel1[45];
CRGB ledsChannel2[69];

CorsairLightingFirmware firmware = corsairLT100Firmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 45);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 69);
  ledController.addLEDs(0, ledsChannel1,27);
  ledController.addLEDs(1, ledsChannel2, 27);
  ledController.onUpdateHook(0, []() {
    CLP::SegmentScaling segments[1] = {{27, 45}};
    CLP::scaleSegments(&ledController, 0, segments, 1 );
  });
  ledController.onUpdateHook(1, []() {
     CLP::SegmentScaling segments[1] = {{27, 69}};
    CLP::scaleSegments(&ledController, 1, segments, 1 );
  });
}

void loop() {
  cHID.update();

  if (ledController.updateLEDs()) {
    FastLED.show();
  }
}
```

kdbhu, 2020-12-25:
All library up to date (fresh install)

Legion2, 2020-12-25:
An L100 only supports one channel (`addLEDs` once). but you can have 4 towers connected to this channel

kdbhu, 2020-12-25:
Thanks, but i have 2 led strip, and i cant connect together them.
Can u help to solve this?

Legion2, 2020-12-25:
try this, it uses a single led array and tell FastLED to use an offset in the array
```
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_1 2
#define DATA_PIN_2 3

CRGB ledsChannel[114];

CorsairLightingFirmware firmware = corsairLT100Firmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
  FastLED.addLeds<WS2812B, DATA_PIN_1, GRB>(ledsChannel, 0,45);
  FastLED.addLeds<WS2812B, DATA_PIN_2, GRB>(ledsChannel, 45, 69);
  ledController.addLEDs(0, ledsChannel,54);
  ledController.onUpdateHook(0, []() {
    CLP::SegmentScaling segments[2] = {{27, 45},{27, 69}};
    CLP::scaleSegments(&ledController, 0, segments, 2);
  });
}

void loop() {
  cHID.update();

  if (ledController.updateLEDs()) {
    FastLED.show();
  }
}

```

kdbhu, 2020-12-25:
Thank you! That's work well.

closed 2020-12-25 by Legion2