Legion2/CorsairLightingProtocol #208: ATmega32u2 Support

cinderblock, 2020-12-28:
I just found your project. I'm very interested in using it.

However, I have some boards based on an ATmega32u2, a slightly simplified version of the ATmega32u4, that I'd like to use.

I'm wondering what special features of the ATmega32u4 you might have used that might not exist on the u2. I'm personally rather familiar with both of these processors and the only 2 major differences I know about are the existence of an ADC (and internal temperature sensor) and the 3-phase PWM output.

If I don't intend to use the internal temperature sensor, do you foresee any issues with using the ATmega32u2 instead?

Thanks!

Legion2, 2020-12-28:
If these are the only differences it should be possible to use the ATmega32u2 as well. I didn't list it currently, because I didn't come across any board that uses an ATmega32u2 chip.

cinderblock, 2020-12-28:
Yeah, the main reason I'm thinking of this is that I'm planning to make a custom board and the ATmega32u2 is available in a smaller package. I happen to have other boards to test with as well.

Thanks for the reply. I'll close this for now. When I try it, and especially if I have issues, I'll report back here, assuming I remember!

closed 2020-12-28 by cinderblock