Legion2/CorsairLightingProtocol #209: Issue Controlling Non-Addressable LEDs

Murdy007, 2020-12-29:
I've just been trying to control some 5050 LEDs using the 'NonAddressable' example for an Arduino Leonardo.
There seems to be an issue with the color of the LEDs - when I set the color to R:255 G:0 B:0 in iCUE, the LEDs show green and blue at full brightness with a dim red color, and when I set it to R:0 G:0 B:255, the LEDs show red and green at full brightness but blue is very dim.
I haven't modified the example, but if you had any ideas as to why this is happening your help is much appreciated.
Thank you.

Legion2, 2020-12-29:
The wiring of your LEDs is inverted. What LEDs do you use, depending on that you have to options: change the wiring or invert the arduino output signal.

Murdy007, 2020-12-29:
I’m wanting to set this up to control the LEDs on a waterblock, and I’m using a strip of 5050 LEDs I bought from China to test it (they are the same LEDs and the same issue occurs on both the waterblock and test strip).
The +12V on the strip is wired into a molex plug, with the R G B wires from the strip connected to the 3, 5, 6 pins on the Arduino.

Legion2, 2020-12-29:
You should not connect the R G B wires directly to the arduino, because the arduino works with 5V (thats why your leds are not off but only dim). You need to use transistors which can handle the 12V and can be controlled by an arduino

Murdy007, 2020-12-29:
I’m assuming wiring it like it shows in this article would work? https://www.makeuseof.com/tag/connect-led-light-strips-arduino/

I’m very new at this and have practically no idea what I’m doing so thank you for helping me.

Legion2, 2020-12-29:
Yes the article looks good, but I think it's much easier to use WS2811/12/12B LED strips, which are also explained in the article. They work with 5V and can directly be controlled by the arduino. When the arduino is connected with USB you should not connect the 5V of a power supply to the arduino pins, because the arduino is powered by USB, only the strip should get the power from the Power supply.

closed 2021-01-21 by stale[bot]