Legion2/CorsairLightingProtocol #210: Re-ordering or transposing fan leds to support non corsair fans

cadrion77, 2021-01-01:
I am a new user to your project...thanks for putting the work into this- pretty cool stuff.  

What I was wondering is if it's possible to re-order or transform led numbering to be able to use a fan other than corsair?  I see that a transform can be down from a ll to strip, so I thought i  may be possible to do this.

Thanks

Legion2, 2021-01-01:
Yes, it is possible to device segments on the iCUE strip and scale them to segments of the physical strip. https://github.com/Legion2/CorsairLightingProtocol/wiki/Scale-separate-segments-of-a-channel

cadrion77, 2021-01-02:
AH... I get it now.  I'm not all that much of a programmer, though I have played with arduino a bit so I can muddle through.  I just recently got hooked on the RGB drug...haha... Thanks for the help.

closed 2021-01-02 by cadrion77