Legion2/CorsairLightingProtocol #211: Support New iCue Commander Core?

JoseArtunedo, 2021-01-11:
Thank you first of all, for your great project, its potential is undeniable.

Will support be added for the new Icue Commander Core?

[image]
[https://www.amazon.es/gp/product/B08G1Q3GZR?psc=1](url)

I was investigating, and I have not seen a trace, but the temperature sensors could be included in the arduino code or it is not possible.

Thank you

Legion2, 2021-01-12:
The link does not work and I can't find any reference for the iCue Commander Core.

kostyamat, 2021-01-14:
Hello.  Please tell me which type of thermistors is suitable for Commander Pro?
Will this work?
€ 0,83 | 5PCS/LOT Thin Film Thermistor ntc 10k 1% 25mm SMD 103 Membrane Temperature Sensor  for 3D Printer Termistor
https://a.aliexpress.com/_BTrsYz

Legion2, 2021-01-14:
10 k ohm Thermistor should work

JoseArtunedo, 2021-01-15:
> El enlace no funciona y no puedo encontrar ninguna referencia para el iCue Commander Core.

I do not know why the link I sent cannot be displayed. I leave you another.

Also adding the diagram of connections.

https://www.corsair.com/es/es/Categor%C3%ADas/Productos/Accesorios-%7C-Piezas/Commander-CORE-RGB-&-PWM-Fan-Controller/p/CW-8960071

[image]

JoseArtunedo, 2021-01-15:
> Hola. Dígame qué tipo de termistores es adecuado para Commander Pro.
> esto funcionara?
> € 0,83 | 5 unids / lote termistor de película delgada ntc 10k 1% 25mm SMD 103 Sensor de temperatura de membrana para termistor de impresora 3D
> https://a.aliexpress.com/_BTrsYz

Great this info comes to me, I was also looking for and had the doubt if they were 10K, I had the original commander pro at home, I measured and if they are 10k to work.

Legion2, 2021-01-15:
I think it should be possible to support the Commander Core, now only the USB PID is required, so iCUE will detect it.

JoseArtunedo, 2021-01-15:
I think so, it's basically an all-in-one commander pro + Lighting Node Core. It will simplify everything to the maximum. With this, it is simply all you need to be able to control PWM and lighting from the same place.

Legion2, 2021-01-15:
The features of a commander core can already be realized with this library, just combine the commander pro and the [MultipleFans](https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/MultipleFans/MultipleFans.ino) example.

The USB PID can be obtained from the original device or by guessing it.

JoseArtunedo, 2021-01-15:
I am waiting to receive a Corsair iCUE H150i ELITE CAPELLIX, which contains the Commander Core, if you explain to me how I can obtain or know the USB Pid I will provide it to you. Is this part I do not know how to do it, I have the basic knowledge of arduino and programming

Spegs21, 2021-02-11:
> I am waiting to receive a Corsair iCUE H150i ELITE CAPELLIX, which contains the Commander Core, if you explain to me how I can obtain or know the USB Pid I will provide it to you. Is this part I do not know how to do it, I have the basic knowledge of arduino and programming

Have you found the PID yet? You can use [HWiNFO ](https://www.hwinfo.com/download/) or just device manager to find the PID. In HWiNFO, expand Ports->USB in the device tree. Then expand all of the USB controllers and hubs until you find the Commander Core.  PID is listed in the Hardware ID line. It'll look like:

`Hardware ID:                                                                    USB\VID_1B1C&PID_1B2D`

z0rlac, 2021-02-11:
Using USBDeview, here is the information from my Commander Pro attached to a H150i Capellix

Device Name: CORSAIR iCUE Commander CORE
PID: USB\VID_1B1C&PID_0C1C\<device serial number here>

Legion2, 2021-02-11:
> Device Name: CORSAIR iCUE Commander CORE
PID: USB\VID_1B1C&PID_0C1C<device serial number here>

Looks good, I will try this weekend to create a sketch for the commander core.

Legion2, 2021-02-14:
I tried to use the PID and iCUE tries to use the devices, but it does not show up in iCUE.
Here is an excerpt from the iCUE log:
```
2021-02-14T19:26:22 I cue.dev.hidenumerator: Found device VID 6940 PID 3100 Usage Page 65472 Usage 3072
2021-02-14T19:26:22 I cue.dev.mgr: Candidate device for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "input")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))  not found.
2021-02-14T19:26:22 I cue.dev.mgr: No manifest for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "input")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))
2021-02-14T19:26:22 I cue.dev.mgr: Candidate device for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "input")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))  not found.
2021-02-14T19:26:22 I cue.dev.mgr: No manifest for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "input")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))
2021-02-14T19:26:22 I cue.dev.mgr: Candidate device for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "output")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))  not found.
2021-02-14T19:26:22 I cue.dev.mgr: No manifest for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "output")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))
2021-02-14T19:26:22 I cue.dev.mgr: Candidate device for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "output")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))  not found.
2021-02-14T19:26:22 I cue.dev.mgr: No manifest for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "output")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))
2021-02-14T19:26:22 I cue.hid_device_detector: DeviceInstance(Path: "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}", InstanceId: "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000", ParentInstanceId: "USB\\VID_1B1C&PID_0C1C\\FB66DF55421900F8\u0000") inserted.
2021-02-14T19:26:22 I cue.hid_device_detector: Add open task: "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" with delay 0 ms.
2021-02-14T19:26:22 I cue.hid_device_detector: "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" opened.
2021-02-14T19:26:22 I cue.hid_device_detector: Mark all tasks as finished.
2021-02-14T19:26:22 I cue.devices: Begin attaching to "Commander CORE" (vid=1b1c, pid=c1c) "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" status b= 0
2021-02-14T19:26:22 I cue.devices: End attaching to "Commander CORE" (vid=1b1c, pid=c1c) "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" status a= 0
2021-02-14T19:26:22 I cue.devices: Created new device"Commander CORE" (vid=1b1c, pid=c1c); ready:false
```

@z0rlac can you look into the iCUE logs `CUELogs` and compare them to my logs.

JoseArtunedo, 2021-02-14:
> Intenté usar el PID e iCUE intenta usar los dispositivos, pero no aparece en iCUE.
> Aquí hay un extracto del registro de iCUE:
> 
> ```
> 2021-02-14T19:26:22 I cue.dev.hidenumerator: Found device VID 6940 PID 3100 Usage Page 65472 Usage 3072
> 2021-02-14T19:26:22 I cue.dev.mgr: Candidate device for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "input")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))  not found.
> 2021-02-14T19:26:22 I cue.dev.mgr: No manifest for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "input")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))
> 2021-02-14T19:26:22 I cue.dev.mgr: Candidate device for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "input")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))  not found.
> 2021-02-14T19:26:22 I cue.dev.mgr: No manifest for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "input")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))
> 2021-02-14T19:26:22 I cue.dev.mgr: Candidate device for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "output")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))  not found.
> 2021-02-14T19:26:22 I cue.dev.mgr: No manifest for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "output")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))
> 2021-02-14T19:26:22 I cue.dev.mgr: Candidate device for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "output")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))  not found.
> 2021-02-14T19:26:22 I cue.dev.mgr: No manifest for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000")("hid-direction", "output")("hid-type", "report")("hid-usage", "3072")("hid-usagepage", "65472")("model-name", "Commander CORE")("usb-pid", "3100")("usb-vid", "6940")("vendor", "Corsair"))
> 2021-02-14T19:26:22 I cue.hid_device_detector: DeviceInstance(Path: "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}", InstanceId: "hid\\vid_1b1c&pid_0c1c&mi_02\\7&52bfe76&0&0000", ParentInstanceId: "USB\\VID_1B1C&PID_0C1C\\FB66DF55421900F8\u0000") inserted.
> 2021-02-14T19:26:22 I cue.hid_device_detector: Add open task: "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" with delay 0 ms.
> 2021-02-14T19:26:22 I cue.hid_device_detector: "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" opened.
> 2021-02-14T19:26:22 I cue.hid_device_detector: Mark all tasks as finished.
> 2021-02-14T19:26:22 I cue.devices: Begin attaching to "Commander CORE" (vid=1b1c, pid=c1c) "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" status b= 0
> 2021-02-14T19:26:22 I cue.devices: End attaching to "Commander CORE" (vid=1b1c, pid=c1c) "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#7&52bfe76&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" status a= 0
> 2021-02-14T19:26:22 I cue.devices: Created new device"Commander CORE" (vid=1b1c, pid=c1c); ready:false
> ```
> 
> @ z0rlac, ¿puedes mirar los registros de iCUE `CUELogs`y compararlos con mis registros?

Hello, tell me where I can contact you and I'll pass you the closed beta of iCue. I think it may be because of that.

Legion2, 2021-02-14:
I have iCUE 3.37.140. You can export the logs in the iCUE settings
[image] in the exported zip there is a `CUELogs` folder with the log files.

JoseArtunedo, 2021-02-14:
I have 4.5.134, where I think new data was attached, for the Commader Core. Talk to me here Discord (Jaren # 0618) and if you want to try the code under that version or I will provide you with the closed beta.

Legion2, 2021-02-14:
I have access to the closed beta, but I don't use it, because it does not receive the normal updates.

JoseArtunedo, 2021-02-14:
Ahh oki, then the only thing I can do is try the skets in that version to see if it works for me.

Legion2, 2021-02-14:
I did not modify any sketch, just changed the PID in the device config `AppData\Local\arduino15\packages\Legion2\hardware\avr\0.3.0\boards.txt`

JoseArtunedo, 2021-02-15:
Same problems you have, I have obtained the same results. The name of the device does not matter what it is, the important thing is the pid and it seems that it is not **0x0c1c** I will continue investigating.

z0rlac, 2021-02-15:
Is there anything I can help with? I exported my system info and logs. Under the "Corsair Devices" in SystemInfo.txt I have these two entries:

Device name : Corsair CORSAIR iCUE Commander CORE (serial number removed)
Status : DeviceConnected (0x1)
Working speed : High Speed (0x2)
Connected to : USB xHCI Compliant Host Controller ? Root Hub ? Port 5
USB VID : 0x1b1c
USB PID : 0xc1c
Hub driver : Standard, USB 3.0
Device supports USB 3.0 : 
Device support flags : 
Port protocol support : 

--

Device name :                                                  
Status : DeviceConnected (0x1)
Working speed : Full Speed (0x1)
Connected to : USB xHCI Compliant Host Controller ? Root Hub ? Port 5
USB VID : 0x1b1c
USB PID : 0x1c05
Hub driver : Standard, USB 3.0
Device supports USB 3.0 : 
Device support flags : 
Port protocol support :

JoseArtunedo, 2021-02-15:
> ¿Hay algo en lo que pueda ayudar? Exporté la información y los registros de mi sistema. Debajo de "Dispositivos Corsair" en SystemInfo.txt tengo estas dos entradas:
> 
> Nombre del dispositivo: Corsair CORSAIR iCUE Commander CORE (número de serie eliminado)
> Estado: Dispositivo conectado (0x1)
> Velocidad de trabajo: Alta velocidad (0x2)
> Conectado a: ¿Controlador de host compatible con USB xHCI? Concentrador raíz ? Puerto 5
> USB VID: 0x1b1c
> USB PID: 0xc1c
> Controlador de concentrador: Estándar, USB 3.0
> Dispositivo compatible con USB 3.0: Indicadores de compatibilidad de dispositivo:
> Soporte de
> protocolo de puerto:
> 
> -
> 
> Nombre del dispositivo: 
> Estado: Dispositivo conectado (0x1)
> Velocidad de trabajo: Velocidad máxima (0x1)
> Conectado a: ¿Controlador de host compatible con USB xHCI? Concentrador raíz ? Puerto 5
> USB VID: 0x1b1c
> USB PID: 0x1c05
> Controlador de concentrador: Estándar, USB 3.0
> Dispositivo compatible con USB 3.0: Indicadores de compatibilidad de dispositivo:
> Soporte de
> protocolo de puerto:

**Thanks for sharing. I'm a little lost. Maybe pa @Legion2 Legion2 knows better**

For this device I keep getting this.
> Nombre del dispositivo: Corsair CORSAIR iCUE Commander CORE (número de serie eliminado)
> USB VID: 0x1b1c
> USB PID: 0xc1c

```
2021-02-15 13:42:29.659 I cue.devices: Begin attaching to "Commander CORE" (vid=1b1c, pid=c1c) "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#8&311d48f2&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" status b= 0
2021-02-15 13:42:29.659 I cue.sys_info.win: Info: prop=Manufacturer class=Win32_DiskDrive returned=false
2021-02-15 13:42:29.659 I cue.devices: End attaching to "Commander CORE" (vid=1b1c, pid=c1c) "\\\\?\\hid#vid_1b1c&pid_0c1c&mi_02#8&311d48f2&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}" status a= 0
2021-02-15 13:42:29.659 I cue.devices: Created new device"Commander CORE" (vid=1b1c, pid=c1c); ready:false
2021-02-15 13:42:29.666 I cue.sys_info.win: Info: prop=Size class=Win32_DiskDrive returned=false
2021-02-15 13:42:30.250 I cue.sys_info.win: Info: prop=Caption class=Win32_PnpEntity returned=false
2021-02-15 13:42:30.304 I cue.sys_info.win: Info: prop=DeviceID class=Win32_PnpEntity returned=false
```
Also try this device, and it does recognize it
Nombre del dispositivo: HX750i (Power supply)
> USB VID: 0x1b1c
> USB PID: 0x1c05
[image]

z0rlac, 2021-02-15:
Ah yes, I do have HX750i installed and connected via USB, so we can be sure that is not the PID for the Commander Core

cnn123666, 2022-02-11:
Device Information                                                              
Device Manufacturer:                                                            Corsair
Product Name:                                                                   CORSAIR iCUE Commander CORE
Serial Number:                                                                  a20a30705ecb3abad56b36f52091005f
USB Version Supported:                                                          2.00
USB Device Speed:                                                               USB 2.0 High-speed
Driver Description:                                                             USB Composite Device
Hardware ID:                                                                    USB\VID_1B1C&PID_0C1C

Driver Information                                                              
Driver Manufacturer:                                                            (标准 USB 主控制器)
Driver Description:                                                             USB Composite Device
Driver Provider:                                                                Microsoft
Driver Version:                                                                 10.0.19041.488
Driver Date:                                                                    21-Jun-2006
DeviceInstanceId                                                                USB\VID_1B1C&PID_0C1C\A20A30705ECB3ABAD56B36F52091005F
Location Paths                                                                  PCIROOT(0)#PCI(0801)#PCI(0003)#USBROOT(0)#USB(1)
[image]

Spegs21, 2022-02-16:
I added the information in my latest PR. It still won't show; I'm guessing we are missing new commands that iCue is sending to it or it doesn't like the device ID. 

I'd like to get the info for a Commander Core XT as it doesn't have the integrated pump controller and would probably be easier to support.

Spegs21, 2022-02-24:
If I try to configure a Commander Pro by changing the TinyUSB sketch from CORSAIR_LIGHTING_NODE_PRO to CORSAIR_COMMANDER_PRO it doesn't show in iCue with a Pico. I still need to try modifying the command pro sketch for TinyUSB to see if it work. This leads me to believe that simply changing PIDs is not enough to get something to show up. @Legion2 when you tested this [here](https://github.com/Legion2/CorsairLightingProtocol/issues/211#issuecomment-778838949) did you use the commander pro sketch?

cnn123666, 2022-04-09:
[image]
[image]
The Commander core and Commander coreXT
I think Commander  core and Commander coreXT have the same hardware, and Commander  core can write to Commander  coreXT firmware.
Commander Core has a cold head interface, which contains PWM, temperature probe (monitoring water temperature), light, ID (used to identify the model of water cooling radiator)
Commander Core XT has a small board attached to it, which brings out the temperature probe, light, and ID.

JoseArtunedo, 2022-04-09:
These images are very interesting to see how these products are designed, although it is not necessary (in quotes)

Here the important thing is to know the ID of each device (As recognized by windows and the iCue software) So that when using the arduino, said ID is the same and iCue can be fooled and access the functionalities of these devices, without being ID does not appear in iCue.

If it is true that when we see these images we can see what it does, and thus add the functions when creating an emulate a project.

At the moment the Commander Core ID has been obtained; but could not "emulate so that it appears in iCue"
I don't know if the Commander Core XT ID is known; could you share it to try (at the beginning of this question you can see how to get it)
For Commander Pro, there is already a project that works and is recognizable in iCue

Spegs21, 2022-04-09:
> These images are very interesting to see how these products are designed, although it is not necessary (in quotes)
> 
> Here the important thing is to know the ID of each device (As recognized by windows and the iCue software) So that when using the arduino, said ID is the same and iCue can be fooled and access the functionalities of these devices, without that ID, the device will not appear in iCue.
> 
> If it is true that when we see these images we can see what it does, and thus add the functions when creating an emulate a project.
> 
> At the moment the Commander Core ID has been obtained; but could not "emulate so that it appears in iCue" I don't know if the Commander Core XT ID is known; could you share it to try (at the beginning of this question you can see how to get it) For Commander Pro, there is already a project that works and is recognizable in iCue

Unfortunately, it's a bit more complicated than that. The protocol is completely different for these devices. I've used Wireshark to sniff the HID packets to and from my Command Core and it doesn't look anything the controller implementation in the repo. I've been able to decode some of the commands and values but I haven't had much time to work on it.

Additionally, these devices use a microcontroller with USB High-Speed capability, now confirmed thanks to @cnn123666 . This is evident in the packet size I posted above as well as Full-Speed maxes out at 64 bytes per HID packet for interrupt transfers. No Arduinos and few third-party boards support HS. I think we can still get most of the functionality because it doesn't seem like it uses much of the packet but a lot of work still needs done on figuring out the protocol.

cnn123666, 2022-04-10:
@JoseArtunedo Is this what you want?
Device Information                                                              
Device Manufacturer:                                                            Corsair
Product Name:                                                                   CORSAIR iCUE COMMANDER CORE XT
Serial Number:                                                                  d10f20a08ac184aa6c3a0c065091005f
USB Version Supported:                                                          2.00
USB Device Speed:                                                               USB 2.0 High-speed
Driver Description:                                                             USB Composite Device
Hardware ID:                                                                    USB\VID_1B1C&PID_0C2A

Driver Information                                                              
Driver Manufacturer:                                                            (Standard USB Host Controller)
Driver Description:                                                             USB Composite Device
Driver Provider:                                                                Microsoft
Driver Version:                                                                 10.0.19041.488
Driver Date:                                                                    21-Jun-2006
DeviceInstanceId                                                                USB\VID_1B1C&PID_0C2A\D10F20A08AC184AA6C3A0C065091005F
Location Paths                                                                  PCIROOT(0)#PCI(1A00)#USBROOT(0)#USB(1)#USB(6)#USB(3)

cnn123666, 2022-04-10:
This is used when ICUE is not started (CORSAIR ICUE COMMANDER CORE XT)
Device Information                                                              
Device Manufacturer:                                                            Corsair
Product Name:                                                                   Corsair, PID=1D0B
Serial Number:                                                                  -
USB Version Supported:                                                          2.00
USB Device Speed:                                                               USB 2.0 High-speed
Driver Description:                                                             USB Composite Device
Hardware ID:                                                                    USB\VID_1B1C&PID_0C2A

Driver Information                                                              
Driver Manufacturer:                                                            (Standard USB Host Controller)
Driver Description:                                                             USB Composite Device
Driver Provider:                                                                Microsoft
Driver Version:                                                                 10.0.19041.488
Driver Date:                                                                    21-Jun-2006
DeviceInstanceId                                                                USB\VID_1B1C&PID_0C2A\D10F20A08AC184AA6C3A0C065091005F
Location Paths                                                                  PCIROOT(0)#PCI(1A00)#USBROOT(0)#USB(1)#USB(6)#USB(3)

cnn123666, 2022-04-10:
This is used when ICUE is not started (CORSAIR ICUE COMMANDER CORE )
Device Information                                                              
Device Manufacturer:                                                            Corsair
Product Name:                                                                   Corsair, PID=1D0B
Serial Number:                                                                  -
USB Version Supported:                                                          2.00
USB Device Speed:                                                               USB 2.0 High-speed
Driver Description:                                                             USB Composite Device
Hardware ID:                                                                    USB\VID_1B1C&PID_0C1C

Driver Information                                                              
Driver Manufacturer:                                                            (Standard USB Host Controller)
Driver Description:                                                             USB Composite Device
Driver Provider:                                                                Microsoft
Driver Version:                                                                 10.0.19041.488
Driver Date:                                                                    21-Jun-2006
DeviceInstanceId                                                                USB\VID_1B1C&PID_0C1C\E00800207C8A5ABA2DD366F53091005F
Location Paths                                                                  PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(3)