Legion2/CorsairLightingProtocol #213: Pro Micro Hardware Lighting Bug

LiftKid, 2021-02-15:
**Describe the bug**
When using and arduino Pro Micro clone, hardware lighting will not work.

**To Reproduce**
Steps to reproduce the behavior:
Upload the code to a Pro Micro sparkfun clone (https://bit.ly/3pu6ya4) and try to set a hardware lighting setup in icue. It will just do a random effect or turn off. I have tried on two boards with two programs and both were buggy. For example, my friends board running the lighting node pro sketch just turned off the LEDs for hardware lighting whereas my board running the LS100 sketch will just be set to white instead of my set hardware lighting mode.

**Expected behavior**
I expected the LEDs to switch to the set hardware lighting mode.

**Screenshots**
I dont have any.

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Sparkfun Pro Micro Clone (https://bit.ly/3pu6ya4)
 - Device: LS100 and Lighting Node Pro
 - Version of IDE: 1.8.13
 - Version of CLP: 0.14.1
 - Version of CLP Boards: 0.3.0
 - Sketch: LS100 and Lighting Node Pro

**Code changes**
None

Legion2, 2021-02-15:
@LiftKid Can you make a screenshot of your hardware lighting config in iCUE.

LiftKid, 2021-02-15:
> @LiftKid Can you make a screenshot of your hardware lighting config in iCUE.

[image]

Legion2, 2021-02-15:
You know that with the LS100 the channels in iCUE are interchanged, so Channel 1 is Channel 2? Have you set the hardware lighting for the correct channel?

LiftKid, 2021-02-15:
Yeah its definitely set correct because the preview works.

Legion2, 2021-02-15:
can you try and use the latest version of the CLP library (0.14.2)

LiftKid, 2021-02-15:
I updated to that library and re flashed the arduino. It still gets stuck in white instead of switching to hardware lighting.

Legion2, 2021-02-15:
You use the LS100 sketch without any code changes?

LiftKid, 2021-02-15:
Yeah I just load it straight from the examples page.

Legion2, 2021-02-15:
So when you close/stop iCUE, the arduino goes into hardware lighting mode, but the lighting effect is not the one you configured in iCUE before?

LiftKid, 2021-02-15:
If I use task manager to force close Icue the strips stay white and then turn off after 30 seconds. But if I shut my computer down they just stay white and never shut down. Just to make sure however I will try leaving my computer off for a minute to see if they eventually turn off.

Legion2, 2021-02-15:
> turn off after 30 seconds

this is the hardware lighting mode timeout, when the arduino gets no command from iCUE in 30 seconds it switches into the hardware lighting mode

LiftKid, 2021-02-15:
Yeah. It does seem that it's able to get into hardware lighting mode, just not during shutdown. I turned off my computer for 1 minute and it did not have the same effect as closing icue forcefully. (when it should)

Legion2, 2021-02-15:
what iCUE version do you use?

LiftKid, 2021-02-15:
3.37.140, the newest.

Legion2, 2021-02-15:
I think this is then a problem of iCUE. I have also this version of iCUE, but my devices go into the correct hardware lighting mode when I stop iCUE.

LiftKid, 2021-02-15:
Ok thanks, I hope that it will just eventually fix itself.

closed 2021-02-15 by LiftKid