Legion2/CorsairLightingProtocol #214: SAMD Architecture Support

Spegs21, 2021-02-15:
I see that there have been a lot of questions regarding support of other platform looking through the readme but one that hasn't come up is the SAMD architecture, specifically the SAMD21 chip.

I think this should meet the raw HID requirements that many of the other chips do not. One drawback I see is the lack of an EEPROM but this could be mitigated with a flash storage library or external EEPROM.

I've made some preliminary changes to the library but don't have anything that compiles yet.

Legion2, 2021-02-15:
Can you provide a list of arduino board names, which I can use to test SAMD. I only need the names to setup the compiler in arduino IDE.

Spegs21, 2021-02-15:
Sure,

**Official Arduino Boards:**

- Arduino Zero
- Arduino MKR Zero
- Arduino Nano 33 IoT
- Arduino MKR WAN 1300
- Arduino MKR WAN 1310
- Arduino MKR NB 1500
- Arduino MKR1000
- Arduino MKR WiFi 1010
- Arduino MKR GSM 1400
- Arduino MKR Vidor 4000

**Adafruit Boards:**

- Trinket M0
- Feather M0
- Metro M0
- Gemma M0
- ItsyBitsy M0

Legion2, 2021-02-15:
OK the EEPROM api has to be abstracted from the code, so it can be replaced when needed.

Spegs21, 2021-02-16:
Is that the only issue you've found? I edit the HID files to allow ARDUIINO_ARCH_SAMD and the compiler complains about lack of USB_Send and TRANSFER_RELEASE depending on which board I select.

Legion2, 2021-02-16:
https://www.arduino.cc/en/Reference/HID here only 32u4 based board, a Due and a Zero are supported

Spegs21, 2021-02-16:
Have you enabled your #if blocks in RawHID.h/.cpp by changing:
`#if defined(ARDUINO_ARCH_AVR)` to  `#if defined(ARDUINO_ARCH_AVR) || #if defined(ARDUINO_ARCH_SAMD)`

I'm trying to compile for Arduino Zero (Native USB Port).

Legion2, 2021-02-16:
no I didn't started coding, can you share your changes and open a PR?

Spegs21, 2021-02-17:
I added the PR. I don't have much experience with the USB API and HID libraries so I don't know how much help I'll be with this.

closed 2022-01-31 by Legion2