Legion2/CorsairLightingProtocol #216: Build with VS Code and PlatformIO

thilbig, 2021-02-17:
Hi,
I open the project with VS Code and PlatformIO PlugIn. I can compile and upload it to my Arduino Leonardo. But the device isn't shown in iCUE software.
When I build it with ArduinoIDE then all works as described. Is it possible to port it to PlatformIO ? It's more easier to use with VS Code and PlatformIO than in ArduinoIDE.

Legion2, 2021-02-17:
I don't know how to support the custom device setting which can be done in arduino IDE using the boards definition

thilbig, 2021-02-17:
OK. Then I will give it a try and when it works I will do a pull request. Is that ok for you?

closed 2021-02-17 by thilbig

Legion2, 2021-02-17:
Yes, pull requests are welcome.

zastrixarundell, 2023-11-19:
I'll see to make a PR for the README but here's the required PlatformIO thing needed:
```ini
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board_build.core = earlephilhower
board = pico
framework = arduino
build_flags = -DUSE_TINYUSB
upload_protocol = picotool
lib_deps =
	legion2/Corsair Lighting Protocol@^0.15.0
	adafruit/Adafruit TinyUSB Library@^2.2.6
```

Successfully built on my PC (which wouldn't otherwise) and uploaded on an unofficial board.