Legion2/CorsairLightingProtocol #218: CorsairLightingProtocol/examples/MultipleFans/MultipleFans.ino Error

MinadaLakmira, 2021-03-08:
CorsairLightingProtocol/examples/MultipleFans/MultipleFans.ino

This Code only control 8 leds on Corsair LL, How to setup it for all 16 leds ?

robbo247, 2021-03-08:
i dont know if it works the same but you can edit the number of leds in the
ino file directly then upload it too your arduino

Legion2, 2021-03-08:
@MinadaLakmira you have to change the line `#define NUMBER_OF_LEDS_PER_FAN 8` to 16

MinadaLakmira, 2021-03-08:
No problem, I found the answer 👍👍-------- Original message --------From: Robbo247 <notifications@github.com>Date: Mon, 8 Mar 2021, 11:59To: Legion2/CorsairLightingProtocol <CorsairLightingProtocol@noreply.github.com>Cc: MinadaLakmira <pminadalakmiragaming@gmail.com>, Author <author@noreply.github.com>Subject: Re: [Legion2/CorsairLightingProtocol] CorsairLightingProtocol/examples/MultipleFans/MultipleFans.ino Error  (#218)
i dont know if it works the same but you can edit the number of leds in the
ino file directly then upload it too your arduino

closed 2021-03-08 by Legion2

robbo247, 2021-03-08:
Are people actually using corsairs fans etc on this project just too bypass
a genuine node etc ,i thought this project was aimed more as a custom non
brand lighting workaround