Legion2/CorsairLightingProtocol #225: Multiple Fans Configuration Not Working

hahagu, 2021-04-06:
**Describe the problem**
The code below introduces flicker, as well as spontaneous turning off of the LED strips.
The 5v lane and the GND is connected to a sata power cable's 5v and GND for power from the motherboard itself.

**Circuit Design**
[image]

**Code**
`/*
   Copyright 2019 Leon Kiefer

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

	   http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.
*/
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define LED_BTN_PIN 10

#define DATA_CHANNEL_A1 2
#define DATA_CHANNEL_A2 3
#define DATA_CHANNEL_A3 4
#define DATA_CHANNEL_A4 5

#define DATA_CHANNEL_B1 16
#define DATA_CHANNEL_B2 14 
#define DATA_CHANNEL_B3 15
#define DATA_CHANNEL_B4 A0

#define CHANNEL_OFFSET_A 16
#define CHANNEL_OFFSET_B 16

CRGB ledChannel1[64];
CRGB ledChannel2[64];

CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

boolean ledBtnOff = false;

int ledBtnReading;
int prevLedBtnReading = LOW;

long time = 0;
long debounce = 500;

void setup() {
  CLP::disableBuildInLEDs();
  pinMode(LED_BTN_PIN, INPUT);
  
  // Channel A                                      Channel           Offset         Number of LEDs
  FastLED.addLeds<WS2812B, DATA_CHANNEL_A1, GRB>(ledChannel1, CHANNEL_OFFSET_A * 0, CHANNEL_OFFSET_A);
  FastLED.addLeds<WS2812B, DATA_CHANNEL_A2, GRB>(ledChannel1, CHANNEL_OFFSET_A * 1, CHANNEL_OFFSET_A);
  FastLED.addLeds<WS2812B, DATA_CHANNEL_A3, GRB>(ledChannel1, CHANNEL_OFFSET_A * 2, CHANNEL_OFFSET_A);
  FastLED.addLeds<WS2812B, DATA_CHANNEL_A4, GRB>(ledChannel1, CHANNEL_OFFSET_A * 3, CHANNEL_OFFSET_A);

  // Channel B                                      Channel           Offset         Number of LEDs
  FastLED.addLeds<WS2812B, DATA_CHANNEL_B1, GRB>(ledChannel2, CHANNEL_OFFSET_B * 0, CHANNEL_OFFSET_B);
  FastLED.addLeds<WS2812B, DATA_CHANNEL_B2, GRB>(ledChannel2, CHANNEL_OFFSET_B * 1, CHANNEL_OFFSET_B);
  FastLED.addLeds<WS2812B, DATA_CHANNEL_B3, GRB>(ledChannel2, CHANNEL_OFFSET_B * 2, CHANNEL_OFFSET_B);
  FastLED.addLeds<WS2812B, DATA_CHANNEL_B4, GRB>(ledChannel2, CHANNEL_OFFSET_B * 3, CHANNEL_OFFSET_B);
	
	ledController.addLEDs(0, ledChannel1, 64);
	ledController.addLEDs(1, ledChannel2, 64);
}

void loop() {
  ledBtnReading = digitalRead(LED_BTN_PIN);
  if (ledBtnReading == HIGH && prevLedBtnReading == LOW && millis() - time > debounce) {
    ledBtnOff = !ledBtnOff;
    time = millis();
  }

  cHID.update();

  if (ledController.updateLEDs()) {
    if (ledBtnOff) {
      FastLED.clear(true);
    }
    FastLED.show();
  }
  prevLedBtnReading = ledBtnReading;
}`

Legion2, 2021-04-06:
Have you tested the hardware wiriwing with another FastLED sketch, so we can make sure it is the sketch which causes the problems and not he hardware setup? For example use a simple moving rainbow effect to test the hardware setup.

Also, which effects have you configured in iCUE?

hahagu, 2021-04-06:
Just a simple rainbow for all, including hardware effects.
The hardware pins are not the problem. I've checked them.
I suspect power or just arduino being overloaded, but I can't really be sure.

Legion2, 2021-04-06:
2x 64 LEDs should not overload the arduino. It can handle a minimum of 200 LEDs with iCUE. Maybe there is a problem with the button which turns of the leds?

hahagu, 2021-04-06:
The button isn't the issue for sure, although the clear function sometimes doesn't want to work for some reason.

Legion2, 2021-04-06:
Can you reproduce the effect with only a single 16 LEDs strip configured in the sketch?

hahagu, 2021-04-06:
With every other lines disabled except a single channel with a single fastled strip, it works fine.

hahagu, 2021-04-07:
It doesn't flicker per say, it just turns off randomly and turns back on randomly.
It doesn't happen repeatedly, only a few times after I connect several strips together.

Legion2, 2021-04-07:
You have used an external pull down resistor for the button, most arduino digital pins have a built-in pull up resistor, which can be activated with `pinMode(x, INPUT_PULLUP)`. Maybe the button turns on and off the led strips, because I can not think of something else that randomly turns all the leds on and off.

hahagu, 2021-04-07:
With some tweaking, I got a channel up and running.
The second channel now supports all 4 strips.
I may have to try resoldering to see if the channels I've soldered makes good contact, and add some diodes to prevent reverse current.
I also have a suspicion that the pro micro that I'm using may be damaged, so that would need checking as well.
I will report back once I get some more things working.

hahagu, 2021-04-08:
[image]

Redesigned the circuit, now far more stable.

However, the off button doesn't seem to be working.
FastLED.clear(true);
maybe doesn't blank the LEDs?

hahagu, 2021-04-08:
`Sketch uses 28618 bytes (99%) of program storage space. Maximum is 28672 bytes.`
Very very close to full.. phew 😅

closed 2021-04-08 by hahagu