Legion2/CorsairLightingProtocol #228: cannot controll more than 223 leds per channel

JurgenKuyper, 2021-04-20:
**Describe the problem**
if I define more than 223 leds in the CRGBArray Icue doesn't detect my controller anymore.
A clear and concise description of your problem.

**Screenshots**
[image](https://user-images.githubusercontent.com/45831568/115446401-74a74980-a217-11eb-825b-7ad3862a64f0.png)

**System (please complete the following information):**
 - OS:Windows 10
 - Board: Arduino Leonardo
 - Device:  Lighting Node PRO
 - Version of IDE: Arduino IDE 1.8.13
 - Version of CLP: 0.14.2
 - Version of CLP Boards:  0.3.1
 - Sketch: RepeatAndScale_customized

**Code changes**
I altered some of the code to accommodate my needs
[RepeatAndScale.zip](https://github.com/Legion2/CorsairLightingProtocol/files/6345915/RepeatAndScale.zip)

Legion2, 2021-04-20:
if you don't use the second channel you can remove it, then you should be able to configure more leds for the first channel.

You should also combine the `ledController.onUpdateHook(0,...` for the first channel into one

JurgenKuyper, 2021-04-21:
> if you don't use the second channel you can remove it, then you should be able to configure more leds for the first channel.
> 
> You should also combine the `ledController.onUpdateHook(0,...` for the first channel into one
thank you, that works, just wondering, why is the amount of leds if there are 2 channels limited to 223?

Legion2, 2021-04-21:
The total amount of leds of both channels together is limited by the memory size of the arduino

closed 2021-04-21 by Legion2