Legion2/CorsairLightingProtocol #233: How Do i wire commander pro using this CorsairLightingProtocol

jiveshd83, 2021-05-20:
I am having some problem how to wire for commander pro and where does fans will go, where will leds, where will temperature sensor i am having a lot of trouble in it

Legion2, 2021-05-20:
In the `CommanderPRO.ino` you can see which pins are used. In the [Wiki](https://github.com/Legion2/CorsairLightingProtocol/wiki/Commander-PRO) you can find more information. However, to control fans and temperature sensors, additional components are required. The hardware wiring is out of scope of this project, you can find more information about PWM fans and temperature sensors in the internet.

jiveshd83, 2021-05-20:
ok so you mean only i can control the pwm and not be able to monitor them, Also tell me for thermal sensor where will go positive and where will negative

jiveshd83, 2021-05-20:
and what additional components are required?

jiveshd83, 2021-05-20:
Thanks Brother Lots of love From India ❤❤

closed 2021-05-20 by Legion2