Legion2/CorsairLightingProtocol #237: Color Accuracy

Tocs13, 2021-06-15:
I'm not sure if this is something that can be improved or not, but when showing white it is more blue than white. 

I'm running WS2812B LED strips using a Leonardo Tiny Atmega32U4 as a LS100 controller using the video feed in icue. The output seems to be biased towards blue. 
This could be an issue with icue or maybe color accuracy within the code? 

Is there something that could be tweaked or calibrated that could fix this?

 - OS: Windows 10
 - Board: Arduino Leonardo tiny [link](https://www.jaycar.com.au/duinotech-leonardo-tiny-atmega32u4-development-board/p/XC4431)
 - Version of IDE: Arduino IDE 1.8.15 and V2.0
 - Version of CLP: 0.14.2
 - Sketch: LS100

I can upload some control photos later show back wall and monitor with all white , blue, red, green, etc if it will help to show how the LED represent each color?

I'm really impressed with it, so a big thanks for all the effort everyone has put into it already.

Legion2, 2021-06-15:
It is a common problem for cheap LEDs, however you can correct it using some settings in arduino. See this [FastLED-Color-Correction](https://github.com/FastLED/FastLED/wiki/FastLED-Color-Correction) guide. You need to call the correction function in the arduino `setup` function.

Tocs13, 2021-06-15:
Thanks Legion. I found that .setTemperature (Tungsten40W) was required to get white close to white as the color correction didn't seem to improve it at all.

closed 2021-06-15 by Legion2