Legion2/CorsairLightingProtocol #238: Can´t get higher ammount of LED running

GitHubsAde, 2021-06-22:
**Describe the problem**

I managed to get the Commander Pro show up in Icue by adding the Board specs in the boars.txt (attatched) by my selfe ( Icouldn´t find it in the documents). Now i get the problem, when i change the CHANNEL_LED_COUNT Value to match the LED strips I am using the commander pro does not refresh. It starts, displays the Hardware lighting and gets stuck there. the Loop still runs ( i added a single led blink). If i get below 97 for the value it works like a charm (thanks for that btw 🥇 ). I had a similar problem with the node pro with an LED Value above 121. all numbers up to 121 run fine, at 122 it breaks. Did i miss anything/ doing anything wrong? 

I am aware that i could clone the LED strips, but i would like to have just one continous flow.

Thanks for The code and the help

Adrian

**Screenshots**
Not Possible

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Leonardo clone  https://www.amazon.de/gp/product/B07FQBQ4Z6/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
 - Device: Lighting commander PRO
 - Version of IDE:  Arduino IDE 1.8.15
 - Version of CLP: 0.14.2
 - Version of CLP Boards: latest
 - Sketch: CommanderPRO.ino
**Code changes**
Did you changed the code of the library or the sketch?
-CHANNEL_LED_COUNT set to 126 (or every value above 96) to match the 6x 350mm strips with 21 LEDs
-Added a commander Pro Board in the Program Files (x86)\Arduino\hardware\arduino\avr boards.txt 
  
[boards.txt](https://github.com/Legion2/CorsairLightingProtocol/files/6697258/boards.txt)

Legion2, 2021-06-23:
You don't need the board.txt file. Please follow the setup guide in the project Readme. If you only need one led channel, you can remove the other and use more LEDs for it.

GitHubsAde, 2021-06-23:
I now tested it with the setup guide. I still get the same problem. I'll try to delete the second channel as soon as I get home. Do you see any opportunity to use two channels and a higher amount of LEDs?

Legion2, 2021-06-24:
A total of 200 led is not a problem for a Pro Micro, but more LEDs could result in errors.

GitHubsAde, 2021-06-24:
So I finally got time to test it. Removing did solve the Problem 👍 it now runs stable.

closed 2021-07-17 by stale[bot]