Legion2/CorsairLightingProtocol #239: Seedstudio XIAO Support possible?

XOIIO, 2021-06-22:
Hi, I've been up far too long to start digging into the nitty gritty of how the library works to see if it's possible, but it might be above me anyhow.

I have a project where I'm using a seedstudio XIAO with HID-Project and it works great, and I'd love to add some icue integration, however the library lists this board as incompatible.

Just wondering if you could look into if it's possible to add this board to the list, and I'm also wondering if it's possible to have both HID-Project and this running on the same arduino. I don't know if the arduino can emulate multiple HID devices like that, I don't have a compatible board on hand to test unfortunately.

Legion2, 2021-06-23:
We are currently working on support in #214 , but I can't give a release date.

XOIIO, 2021-06-23:
Ok, cool, hopefully it's not too difficult. I'd be happy to beta test early builds to confirm whether or not it works and report back, my little project is complete and hooked up to my pc constantly so it would be pretty quick to test.

Legion2, 2021-07-17:
@XOIIO I'm unable to install the [seedstudio XIAO Boards](https://wiki.seeedstudio.com/Seeed_Arduino_Boards/) so I can't test it.

closed 2021-08-18 by stale[bot]

Spegs21, 2022-01-29:
Should be possible with Legion2/CorsairLightingProtocol#215.