Legion2/CorsairLightingProtocol #246: Pi Pico Support

WanaGo, 2021-08-25:
Hello,

What are the chances that this can be ported or made to work on the Raspberry Pi Pico, with the aim of trying to get support for more than 2 channels. 
Thinking along the lines of the Razer Chroma which can support 6 channels.
It has an Arduino core for it, just wondering if it might be a potential candidate?

Thanks

Spegs21, 2022-01-29:
Should be functional with Legion2/CorsairLightingProtocol#215. I've compiled it and flash it but haven't hooked up a fan yet.

WanaGo, 2022-01-29:
> Should be functional with #215. I've compiled it and flash it to a Pico but haven't hooked up a fan yet.

Nice - But Fan, or Lights, do you mean?

Spegs21, 2022-01-30:
I've compiled the Lighting Node Pro sketch for the Pico and flashed it. Lighting Node Pro is lights only. I haven't tried one that incorporates a fan controller. The new Readme from the PR has instructions to do this. So at least lights should work.

WanaGo, 2022-01-30:
> I've compiled the Lighting Node Pro sketch for the Pico and flashed it. Lighting Node Pro is lights only. I haven't tried one that incorporates a fan controller. The new Readme from the PR has instructions to do this. So at least lights should work.

Cool, yep you just mentioned you haven't hooked a fan up yet, I was wondering if that was a typo and you actually meant you hadnt hooked any lights up yet. Lights only is what I thought too, hence the question about Fans.

cnn123666, 2022-01-30:
> 我已经为 Pico 编译了 Lighting Node Pro 草图并将其闪烁。Lighting Node Pro 只有灯光。我还没有尝试过包含风扇控制器的产品。PR 的新自述文件有执行此操作的说明。所以至少灯光应该可以工作。

How to use other devices such as LS100

cnn123666, 2022-01-30:
> 我已经为 Pico 编译了 Lighting Node Pro 草图并将其闪烁。Lighting Node Pro 只有灯光。我还没有尝试过包含风扇控制器的产品。PR 的新自述文件有执行此操作的说明。所以至少灯光应该可以工作。

Hello, I'm sorry that my English is not very good. I have successfully uploaded TinyUSB sketch to Pico according to your tutorial, and ICUE has successfully identified it, but the lighting channel seems to be abnormal and some of it is stuck.

Spegs21, 2022-01-30:
@cnn123666 Did you modify the sketch for the number of LEDs you are running on the channel? It's set to 96 as an example but this needs to match your specific setup.

cnn123666, 2022-01-31:
@Spegs21 I didn't change the sketch, the refresh rate of the lighting channel is wrong, it seems to be only 15FPS, but the hardware channel is fine

cnn123666, 2022-01-31:
https://user-images.githubusercontent.com/52467395/151735897-6c82bd93-72e6-4463-8bd3-604aedd7eb96.mp4

Spegs21, 2022-01-31:
@cnn123666 Thanks for the video. When you say the hardware channels is fine what do you mean?

cnn123666, 2022-02-01:
@Spegs21 When ICUE is turned off, the light refresh rate is normal

cnn123666, 2022-02-01:
https://user-images.githubusercontent.com/52467395/151973000-1233277b-9ca8-4e8c-a6c5-058de0a4c649.mp4

https://user-images.githubusercontent.com/52467395/151973007-9578073b-f71e-4e1a-ad79-ddf0118fa3ea.mp4

Spegs21, 2022-02-01:
@cnn123666 Ok, that makes sense. I only have a spare half broken SP120 Elite that I can test with but I'll try hooking it up. 

What does your Lighting setup in iCue looks like for the Lighting Node Pro (Pico) on that channel?

cnn123666, 2022-02-02:
[image]
[image]

cnn123666, 2022-02-02:
[image]

cnn123666, 2022-02-02:
@Spegs21 I think it's the sketch

Spegs21, 2022-02-02:
@cnn123666 Yes, I think it is too. You can add `CLP::printFps(5000);` after `FastLED.show();` to get an idea of how often it's refreshing. Make sure you call `Serial.begin(115200);` in `setup()`.

I tested this with the TinyUSB sketch on a Trinket M0 and the refresh rate was pretty low. I stepped through the code with a SWD debugger and it doesn't seem like it's receiving the LED_WRITE_TRIGGER command from iCue as often as needed to keep a smooth refresh rate. We probably need to get @Legion2  involved with this.

Legion2, 2022-02-03:
Can you measure the time it takes to process an USB request, so the total time it takes to receive all bytes from the host and process them and send back all response bytes?

Legion2, 2022-02-03:
I also use iCUE 4

Legion2, 2022-02-03:
Can you print which commands are received

Spegs21, 2022-02-03:
@Legion2 I did a quick ratio for my setup (LNP, 1 channel, 1 8-LED Series Fan) and I'm getting a 1:5 ratio of WRITE_LED_TRIGGER to any other WRITE_LED command. I'll add some debug lines to LEDController.cpp to print out the commands.

Legion2, 2022-02-03:
1:5 should be fine

Spegs21, 2022-02-04:
@Legion2 I implemented a debug macro system which I can open a PR for. Here is the pattern it is receiving:

```
Received command
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	32	00	00	08	00	63	51	38	1D	02	00	00	0C	00	00	00
0x10	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x20	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x30	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
Sent response
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
Received command
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	32	00	00	08	01	00	00	00	00	00	54	AF	FF	00	00	00
0x10	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x20	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x30	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
Sent response
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
Received command
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	32	00	00	08	02	E3	D0	D7	EA	FD	AB	50	00	00	00	00
0x10	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x20	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x30	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
Sent response
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
Received command
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	33	FF	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x10	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x20	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x30	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
Sent response
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
Received command
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	38	00	02	00	00	00	00	00	00	00	00	00	00	00	00	00
0x10	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x20	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
0x30	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
Sent response
>>>>	0x0	0x1	0x2	0x3	0x4	0x5	0x6	0x7	0x8	0x9	0xA	0xB	0xC	0xD	0xE	0xF
0x00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00	00
```

Legion2, 2022-02-06:
This looks completely normal

Legion2, 2022-02-06:
the printFps function must be used in the `if (ledController.updateLEDs())` block

Spegs21, 2022-02-07:
> the printFps function must be used in the `if (ledController.updateLEDs())` block

I know, I was just using to to time the loop with just USB interactions to check if USB communications was the bottleneck.

Legion2, 2022-02-08:
your measurement of the command processing duration is 0ms, is this correct, can you use a better resolution or verify that your measurement method works?

Spegs21, 2022-02-09:
@Legion2 There was a problem with my original timing implementation. I've corrected that and increased the resolution. THe worse I've seen is just over 5ms response time:
```
Received command 32 at 36070285us
Write LED color values
Sent response 00 at 36075376us (5091us response time)
Received command 32 at 36080166us
Write LED color values
Sent response 00 at 36085256us (5090us response time)
Received command 32 at 36094125us
Write LED color values
Sent response 00 at 36095137us (1012us response time)
Received command 33 at 36102129us
Write LED trigger
Sent response 00 at 36104580us (2451us response time)
Received command 38 at 36116532us
Write LED mode
Sent response 00 at 36116888us (356us response time)
```

Legion2, 2022-02-09:
What is the fps value if you use it in the if block

Spegs21, 2022-02-09:
It shows 24.8 fps.

Legion2, 2022-02-09:
I think this is a normal refresh rate for icue, I think this is not the problem for the slow animations

Legion2, 2022-02-10:
With the LS100 iCUE device it should be possible to get 135LEDs per channel or with the commander pro you can get more than 200 per channel if you select QL fans.

Does only the fps decrease or also the usb request duration go up?

Spegs21, 2022-02-11:
@Legion2 With the setup I described (generic LEDs strip, 1.4m + 250mm on both channels) the longest response time was 5.2ms. They were higher in general, minimum of about 3ms.

cnn123666, 2022-02-11:
Use 6 QL fans to see

cnn123666, 2022-02-11:
@Spegs21 the frame rate varies with the number of beads

cnn123666, 2022-02-11:
https://user-images.githubusercontent.com/52467395/153577654-686957cf-c94e-44d9-9199-2d25bfe11acb.mp4

Spegs21, 2022-02-11:
> Use 6 QL fans to see

7.4 fps with this setup

Legion2, 2022-02-12:
Only with Pi Pico or also with other boards?

cnn123666, 2022-02-12:
I only have a Pico board. Why is that? Does USB data processing take too long?

Spegs21, 2022-02-13:
@Legion2 I went back and tested this with the Trinket M0 and performance is identical. I've tried a few things to try to solve this but nothing I've done has made the slightest difference:
- Overclock the Pico to 250Mhz
- Disable interrupts in FastLED
- Increase FASTLED_INTERRUPT_RETRY_COUNT
- Increase the response size to 64 bytes

USB response times seem reasonable with no baseline to compare to. Wireshark shows the response coming through in less than 8ms while the Pico shows it's responding in less than 30us (I'm guessing it's just queueing the transfer until it's ready).

Spegs21, 2022-02-14:
@Legion2 I just tried sharing the IRQ between TinyUSB and FastLED and increasing the TinyUSB interrupt from 1000us to 10000us and neither made any difference either.

I'll implement the histogram and see what we get.

Legion2, 2022-02-14:
During your tests with different number of leds, did you only change the LEDs in icue, or did you increased and decreased them in the sketch and recompiled it? If only changed in icue FastLED always should take the same amount of time to rendered the leds, because it is independent of the number send by icue

cnn123666, 2022-02-14:
Change only in ICUE

cnn123666, 2022-02-14:
[image]

https://user-images.githubusercontent.com/52467395/153854171-bd1292a3-c3fe-4a0b-83e8-4b8877583be4.mp4

Legion2, 2022-02-14:
@Spegs21 can you also include the time it tooks from the end of the loop to the beginning of the next invocation, maybe there are some task running we don't know of

Spegs21, 2022-02-14:
@Legion2 Here are the histograms for 10k loops. Inner loop is the sketch loop and outer loop is the core loop (between the end of the sketch loop and beginning of the next one):

[image]

[image]

Legion2, 2022-02-14:
The idle loop is too fast, it produces to much samples in the histogram can we filter them out or use a logarithmic scale?

Spegs21, 2022-02-14:
@Legion2 @cnn123666 I can't believe I missed this. I had the HID polling interval set to **10ms**...

https://github.com/Legion2/CorsairLightingProtocol/blob/ba2f6e8cae0c34c66885f4f13adc89c8cde512d4/src/CorsairLightingProtocolTinyUSBHID.cpp#L28

Changing the 10 to 1 appears to fix the issue. I'll fix it in a PR.

Spegs21, 2022-02-16:
See #270

Legion2, 2022-02-19:
I release [0.15.0](https://github.com/Legion2/CorsairLightingProtocol/releases/tag/v0.15.0) which includes support for Pi Pico. It will be available in the Arduino IDE in about an hour.

closed 2022-02-19 by Legion2