Legion2/CorsairLightingProtocol #247: Multiple emulated devices

CyberianIce, 2021-09-09:
I was wondering if it's possible to emulate 2nd or 3rd Lighting Node Pro via single Arduino Uno in order to increase number of Independant RGB channels via multiple HID devices on composite USB device? If this is not possible, I am interested in how I can control more than two pieces of 3-pin ARGB fans independently (to control effects on each one)? RGB Fan led hub don't help because it's has six 4-pin terminals... Only solution I am aware would be adding another arduino with different serial number and device ID but that require whole Another arduino...

System:
 - OS: Windows 10
 - Board: Arduino Uno R3 16u2
 - Device: Lighting Node PRO
 - Version of IDE: Arduino IDE 1.8.15
 - Version of CLP: 0.14.2
 - Version of CLP Boards: 0.3.1
 - Sketch: LightingNodePro.ino

I didn't changed anything, this is just conceptual question.

Legion2, 2021-09-09:
Have a look at https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/MultipleFans/MultipleFans.ino. You can control multiple fans with different hardware pins from one channel.

CyberianIce, 2021-09-09:
That will do. I have Arduino Uno 16u2 with HoodLoader2 now, how shall I proceed?

Legion2, 2021-09-09:
You should follow the Arduino Uno guide and only copy the required code from the MultipleFans.ino example to the HoodLoader2UnoMegaController.ino sketch

CyberianIce, 2021-09-10:
Okay, I made it, but I had to change few things in this code. 1st, HID device is not supported, and 2nd Arduino Uno R3 16u2 seems to don't have enough memory for 96 leds per channel, so I reduced number of fans to 4. Later I can add my code to examples repository. Now I have question, since I am using aRGB fans with 16 leds per cahnnel, how I can manage in iCUE to have control over all of them? In order to use them, I had to add configuration like this... But iCUE won't let me add just about anything, like 4x CPU blocks (each CPU waterblock have 16leds so I use them for my fans...) Any known hack for this?

[image]

Legion2, 2021-09-10:
The sketch for the 16u2 should not be modified, so it is independent of the number of leds.
You can scale the led segments of your fans to match that of a corsair fan https://github.com/Legion2/CorsairLightingProtocol/wiki/Scale-separate-segments-of-a-channel. LL fans have 16 leds. Maybe you configure it in icue not as HydroX setup, the you have other options. You can also create other corsair device types which provide more options.

CyberianIce, 2021-09-10:
"The sketch for the 16u2 should not be modified, so it is independent of the number of leds." - Now I am loosing you again. What do you mean by that? I was not able to compile MultipleFans.ino until `CorsairLightingProtocolHID` is replaced with `CorsairLightingProtocolSerial`...

Legion2, 2021-09-10:
For Arduino Uno you need two sketches, one for the ATmega16U2 and one for the ATmega328. `CorsairLightingProtocolSerial` is part of the sketch which is uploaded to the ATmega328. The other sketch (`HoodLoader2CLPBridge.ino`) is uploaded to the ATmega16U2 and should not be modified.

CyberianIce, 2021-09-11:
First I followed complete tutorial to install Lighting Nore Pro on Mega with 16u2 and HoodLoader2 and then I just modifyed and uploaded HoodLoader2UnoMegaController.ino to HoodLoader2 Uno with chunk of code from MultipeFans.ino. Maybe I did something wrong, but it works just as I wanted 🤣 Now I don't know should I touch it, because it works

closed 2021-10-17 by Legion2