Legion2/CorsairLightingProtocol #248: LS100 Does not stay connected.

chanyork, 2021-09-20:
**Describe the problem**
LS100 on Arduino Pro Micro Unstable connection.  First irst time connection after the software upload works few minutes than it will disconnect and jump into Unknown USB Device (Device Descriptor Request Failed).  

Tried different USB Port, different Arduino Pro Micro, different Cable results are the same.  

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Pro Micro
 - Device: LS100
 - Version of IDE: 1.8.15
 - Version of CLP: 0.14.2
 - Version of CLP Boards: [e.g. 0.3.1]
 - Sketch: Scale

**Code changes**
Did you changed the code of the library or the sketch?
/*
   Copyright 2019 Leon Kiefer

   Licensed under the Apache License, Version 2.0 (the "License");
   you may not use this file except in compliance with the License.
   You may obtain a copy of the License at

	   http://www.apache.org/licenses/LICENSE-2.0

   Unless required by applicable law or agreed to in writing, software
   distributed under the License is distributed on an "AS IS" BASIS,
   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
   See the License for the specific language governing permissions and
   limitations under the License.
*/
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

// Hint: The channels of the LS100 are swapped in iCUE, so the first channel in iCUE is here channel 2
#define DATA_PIN_CHANNEL_1 A1
#define DATA_PIN_CHANNEL_2 A0

#define BUTTON_PIN A3

// Hint: The ATmega32U4 does not have enough memory for 135 leds on both channels
// Ultrawide is 54 top and bottom 15 left and right
// Setting up Array

CRGB ledsChannel1[22];
CRGB ledsChannel2[6];

CorsairLightingFirmware firmware = corsairLS100Firmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);

const char mySerialNumber[] PROGMEM = "202B6949A001";
CorsairLightingProtocolHID cHID(&cLP, mySerialNumber);

void setup() {
  pinMode(DATA_PIN_CHANNEL_1, OUTPUT);
  pinMode(DATA_PIN_CHANNEL_2, OUTPUT);  
  pinMode(BUTTON_PIN, INPUT_PULLUP);
  CLP::disableBuildInLEDs();
  byte deviceId[4] = { 0x9A, 0xDA, 0xA7, 0x8F };
  firmware.setDeviceID(deviceId);
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 22);
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 6);
	ledController.addLEDs(0, ledsChannel1, 16);
	ledController.addLEDs(1, ledsChannel2, 4);

//segment front represent corsair rear represent actual led

  ledController.onUpdateHook(0, [](){
    CLP::SegmentScaling segments[] = {{5, 5}, {6, 12}, {5, 5}};
    CLP::scaleSegments(&ledController, 1, segments, 3);
  });
}

void loop() {
	static bool lightingEnabled = true;
	cHID.update();

	if (buttonClicked()) {
		lightingEnabled = !lightingEnabled;
		fill_solid(ledsChannel1, 45, CRGB::Black);
		fill_solid(ledsChannel2, 144, CRGB::Black);
		FastLED.show();
	}

	if (lightingEnabled && ledController.updateLEDs()) {
		FastLED.show();
	}
}

/**
 * Handle button of the LS100. The button is optional.
 *
 * @return true if the button was pressed and then released.
 */
bool buttonClicked() {
	static bool previousState = 1;
	bool state = digitalRead(BUTTON_PIN);
	if (previousState == 0 && state == 1) {
		previousState = state;
		return true;
	}
	previousState = state;
	return false;
}

Legion2, 2021-09-21:
Do you use the button to turn on and off the LEDs, if not remove the code for it, else the code must be updated with the correct number of LEDs.

chanyork, 2021-09-21:
Yes, let me try that.

But I can  not get my device connected now.  keep saying Unknown USB Device (Device Descriptor Request Failed).

chanyork, 2021-09-22:
Tried it, I removed the button code.  And still the same.  the LS100 will be recognize by the USB for about 30 seconds sometime longer and then suddenly become Unrecognized Device.

chanyork, 2021-09-22:
I have a Lighting Node Pro - Arduino, which is running perfectly without any issues.  somehow it seem to be the ls100 only to have the issue.

things I have tried.  Uninstall and reinstall Icue: results is the same.  tried on SparkFun Micro, Tried on Pro Micro, Tried on Leonardo it is all the same.  Show Hidden USB device and delete all hidden USB driver and reconnect.  Disable USB power saving mode.  Nothing seems to be working.

chanyork, 2021-09-22:
Will there be any issue if I am running a custom mega32u4 board?  My Lighting Node Pro already running in it and it is 100% okay.  It is LS100 firmware which I am having issue with.

chanyork, 2021-09-22:
My VID is 0000 and PID is 0002

Legion2, 2021-09-22:
can you send the current code, where you removed the button code.

> But I can not get my device connected now. keep saying Unknown USB Device (Device Descriptor Request Failed).

https://github.com/Legion2/CorsairLightingProtocol/wiki/Troubleshooting

chanyork, 2021-09-23:
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define DATA_PIN_CHANNEL_1 A1 //Channel1
#define DATA_PIN_CHANNEL_2 A0 //Channel0

CRGB ledsChannel1[22];
CRGB ledsChannel2[22];

CorsairLightingFirmware firmware = corsairLS100Firmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
const char mySerialNumber[] PROGMEM = "202B00000002";
CorsairLightingProtocolHID cHID(&cLP, mySerialNumber);

void setup() {
  byte deviceId[4] = { 0x90, 0xDA, 0xA7, 0x90 };
  firmware.setDeviceID(deviceId);

  CLP::disableBuildInLEDs();
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 22);
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 22);
	ledController.addLEDs(0, ledsChannel1, 16);
	ledController.addLEDs(1, ledsChannel2, 16);

  pinMode(DATA_PIN_CHANNEL_1, OUTPUT);
  pinMode(DATA_PIN_CHANNEL_2, OUTPUT);  

//segment front represent corsair rear represent actual led

  ledController.onUpdateHook(0, [](){
    CLP::SegmentScaling segments[] = {{5, 5}, {6, 12}, {5, 5}};
    CLP::scaleSegments(&ledController, 0, segments, 3);
  });
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}

chanyork, 2021-09-23:
Attached is my schematic as well.

[image]

Legion2, 2021-09-23:
Which Power supply do you use?

chanyork, 2021-09-23:
5V-5a to power the led and USB power to power the atmega32U4-MU.  In some
schematic I saw a 1M resistor between the Crystal do I need that?

Legion2, 2021-09-25:
can you reupload the latest sketch and make sure that it is successfully uploaded, in Arduino IDE it should not display an error if successfully uploaded.

chanyork, 2021-09-25:
Hello Legion2, 
Thank you for your help.  I think I probably found the issue to be the USB port connector.  Seems to be a bad quality Female Port which can not keep the connection stable or my trace maybe too think that cause the connection D+/D- very unstable.

closed 2021-09-25 by chanyork