Legion2/CorsairLightingProtocol #249: LS100 Segment and Scaling

chanyork, 2021-09-25:
**Describe the problem**
Need help in the segment and scale coding for LS100.

- Lighting Channel 1 is for Video Sync for ultrawide with 30 on each side and 54 top and bottom, total physical led 168.
Hence scaling is {15, 30}, {54, 54}, {15, 30}, {54, 54}

- Lighting Channel 2 is for mood lamp with physical 20 leds to sync with 3 icue led.
Scaling is {1, 4}, {1,12}, {1,4}

With the c

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Micro
 - Device: LS100
 - Version of IDE: Arduino IDE 1.8.15
 - Version of CLP: 0.14.3
 - Version of CLP Boards: 0.3.1
 - Sketch: Scale

**Code changes**
Did you changed the code of the library or the sketch? yes

`
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

// Hint: The channels of the LS100 are swapped in iCUE, so the first channel in iCUE is here channel 2
#define DATA_PIN_CHANNEL_1 A0
#define DATA_PIN_CHANNEL_2 A1
#define BUTTON_PIN A3

// Hint: The ATmega32U4 does not have enough memory for 135 leds on both channels
// Ultrawide is 54 top and bottom 15 left and right
// Setting up Array

CRGB ledsChannel1[168];
CRGB ledsChannel2[20];

CorsairLightingFirmware firmware = corsairLS100Firmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);

//Declare Serial Number must be unique
const char mySerialNumber[] PROGMEM = "202B00000010";
CorsairLightingProtocolHID cHID(&cLP, mySerialNumber);

void setup() {
//Device ID Setting must be unique
  byte deviceId[4] = { 0x90, 0xDA, 0xA7, 0x10 };
  firmware.setDeviceID(deviceId);
//Define Pinmode
  pinMode(DATA_PIN_CHANNEL_1, OUTPUT);
  pinMode(DATA_PIN_CHANNEL_2, OUTPUT);  
  pinMode(BUTTON_PIN, INPUT_PULLUP);

  CLP::disableBuildInLEDs();
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, 168);
	FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel2, 20);
	ledController.addLEDs(0, ledsChannel2, 3);
  ledController.addLEDs(1, ledsChannel1, 138);

//segment front represent corsair rear represent actual led

  ledController.onUpdateHook(1, [](){
    CLP::SegmentScaling segments[4] = {{15, 30}, {54, 54}, {15, 30}, {54, 54}};
    CLP::scaleSegments(&ledController, 0, segments, 4);
  });

  ledController.onUpdateHook(0, [](){
    CLP::SegmentScaling segments[3] = {{1, 4}, {1, 12}, {1, 4}};
    CLP::scaleSegments(&ledController, 0, segments, 3);
  });

}

void loop() {
	static bool lightingEnabled = true;
	cHID.update();

	if (buttonClicked()) {
		lightingEnabled = !lightingEnabled;
		fill_solid(ledsChannel1, 168, CRGB::Black);
		fill_solid(ledsChannel2, 3, CRGB::Black);
		FastLED.show();
	}

	if (lightingEnabled && ledController.updateLEDs()) {
		FastLED.show();
	}
}

bool buttonClicked() {
	static bool previousState = 1;
	bool state = digitalRead(BUTTON_PIN);
	if (previousState == 0 && state == 1) {
		previousState = state;
		return true;
	}
	previousState = state;
	return false;
}
`

Legion2, 2021-09-25:
`CLP::scaleSegments(&ledController, 0, segments, 4);` should be `CLP::scaleSegments(&ledController, 1, segments, 4);`

chanyork, 2021-09-25:
'''
ledController.onUpdateHook(1, [](){
    CLP::SegmentScaling segments[4] = {{15, 30}, {54, 54}, {15, 30}, {54, 54}};
    CLP::scaleSegments(&ledController, 1, segments, 4);
  });
'''

Noted.  The "onUpdateHook (1" should be 0 right?  What does that do?

Legion2, 2021-09-25:
It is the index of the channel, so it should match the index passed to the scaleSegments

orfin2, 2021-11-23:
Hi, i read "chapter" repeat and scaling and i mean, this is what i need. I solving a problem with lenght of my led strip on my TV. I have 59leds top and botom and 32 left and right sides...total 182leds. I am tryed quick attempt and uploaded example "Repeat or scale LED channels.ino". But after uploading shine only 84leds...if i change value 144 to 182 in this example file, result is the same, still shine only 84 leds. what am I doing wrong.

Legion2, 2021-11-23:
@orfin2 you should use the `AmbientBacklight.ino` and add the segment scalilng codde https://github.com/Legion2/CorsairLightingProtocol/wiki/Scale-separate-segments-of-a-channel

orfin2, 2021-11-23:
i tryed it...but same result, only 84leds shine, I'll continue tomorrow ... I have no idea where the mistake is (maybe too much time with arduino)

orfin2, 2021-11-24:
so...problem was solved, i had damaged led strip and shined only 84 leds. I changed damaged strip and "voila", now shine all 182 leds. But i have one question. Is possible with scaling (segment scaling) code change led strip sequence? In iCUE is possible change directions, but leds start every from bottom of left or right side (shorter strips). I starting from bottom longer TV side (59 leds) - to shorter (32 led) - next top side (59 led) - and last shorter side (32 led). Or is it necessary connect Arduino to LED stripes according to schema in iCUE app?

Legion2, 2021-11-24:
you can apply the `CLP::reverse` function after the segment scaling and then they "start" at the end, which is the bottom stripe.

orfin2, 2021-11-25:
Perfect, now all working on 100%