Legion2/CorsairLightingProtocol #250: Pro micro disconnects

skylersell, 2021-10-16:
**Describe the problem**
I have an issue with my pro micro freezing while in use. If I unplug it and in again it will work sometimes.

[image]

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Clone pro micro 5v
 - Device: ls100
 - Version of IDE: Arduino IDE 1.8.16
 - Version of CLP: 0.14.2
 - Version of CLP Boards: 0.3.1
 - Sketch: ls100
**Code changes**
No code changes

If you need any other info I will try to provide it.

skylersell, 2021-10-17:
When ever its disconnected it shows two red lights along with a green one dose this mean anything all are solid.

Legion2, 2021-10-17:
which arduino clone do you have?

Legion2, 2021-10-21:
how did you connect the arduino to your leds and how are the leds powered?

skylersell, 2021-10-21:
I used my pc power supply and connected the ground and data wire to the leds

Legion2, 2021-11-23:
How many leds do you have connected and configured in iCUE?

closed 2022-01-25 by github-actions[bot]