Legion2/CorsairLightingProtocol #253: Tip to reduce program memory usage / Temperature sensor usage

DanFerras, 2021-11-08:
First, sorry for the bad english, I'm from and my english is soo so...

**Describe the problem**
I'm probably asking too much, but my project is actually trying to control 9 phanteks halo (30 leds each) using a single arduino pro micro, in addition to using two temperature sensors (using a "CommanderPro").
The problem is that I'm running out of program memory, I should probably just try to use a arduino mega, but I would like to know if it's possible to somehow use the pro micro, since I saw in some other answer that I would control even 390 leds with that.
Basically I'm using 6 halos on channel 1 and 3 on channel 2, besides that I'm scaling from 21 to 30, because that's a little better visually (using 2 strips of 15 for each halo also works, but there's a delay between the "strips" and it gets weird).

**error code while compling and upload**
Arduino: 1.8.16 (Windows Store 1.8.51.0) (Windows 10), Board: "CLP SparkFun Pro Micro 5V, 16 MHz, Commander PRO"

In file included from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src/FastLEDController.h:18:0,

                 from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src/CorsairLightingProtocol.h:31,

                 from C:\Users\danil\Documentos\Arduino\CommanderPRO_phanteks2\CommanderPRO_phanteks2.ino:16:

C:\Users\danil\Documentos\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDController.h:18:0,

                 from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDController.cpp:16:

C:\Users\danil\Documentos\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDController.h:18:0,

                 from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDControllerUtils.h:19,

                 from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src\FastLEDControllerUtils.cpp:16:

C:\Users\danil\Documentos\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src\LEDController.h:23:0,

                 from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src\CorsairLightingProtocolController.cpp:18:

C:\Users\danil\Documentos\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src\LEDController.h:23:0,

                 from C:\Users\danil\Documentos\Arduino\libraries\Corsair_Lighting_Protocol\src\LEDController.cpp:16:

C:\Users\danil\Documentos\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

text section exceeds available space in board

Sketch uses 33206 bytes (115%) of program storage space. Maximum is 28672 bytes.

Global variables use 2306 bytes (90%) of dynamic memory, leaving 254 bytes for local variables. Maximum is 2560 bytes.

Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.

Error compiling for board CLP SparkFun Pro Micro 5V, 16 MHz.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: arduino pro micro
 - Device: CommanderPro
 - Version of IDE: Arduino IDE 1.8.16
 - Version of CLP: 0.14.2
 - Version of CLP Boards: 0.3.1
 - Sketch: CommanderPro/MultipleFans/RepeatAndScale

**Code changes**
#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#include "SimpleFanController.h"
#include "ThermistorTemperatureController.h"

#define NUMBER_OF_LEDS_PER_FAN 30

#define DATA_PIN_FAN_1 2
#define DATA_PIN_FAN_2 3
#define DATA_PIN_FAN_3 4
#define DATA_PIN_FAN_4 5
#define DATA_PIN_FAN_5 6
#define DATA_PIN_FAN_6 7
#define DATA_PIN_FAN_7 8
#define DATA_PIN_FAN_8 9
#define DATA_PIN_FAN_9 10

#define TEMP_SENSOR_PIN_1 A0
#define TEMP_SENSOR_PIN_2 A1

#define FAN_UPDATE_RATE 500

#define CHANNEL_LED_COUNT1 180
#define CHANNEL_LED_COUNT2 90

CorsairLightingFirmware firmware = corsairCommanderPROFirmware();
ThermistorTemperatureController temperatureController;
FastLEDController ledController(&temperatureController, true);
SimpleFanController fanController(&temperatureController, FAN_UPDATE_RATE,
								  EEPROM_ADDRESS + ledController.getEEPROMSize());
CorsairLightingProtocolController cLP(&ledController, &temperatureController, &fanController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

CRGB ledsChannel1[CHANNEL_LED_COUNT1];
CRGB ledsChannel2[CHANNEL_LED_COUNT2];

void setup() {
	CLP::disableBuildInLEDs();

  // 6 fans on channel 1
  //                          FAN_PIN           CRGB array            offset             number of leds per fan
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 0, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_2, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 1, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_3, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 2, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_4, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 3, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_5, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 4, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_6, GRB>(ledsChannel1, NUMBER_OF_LEDS_PER_FAN * 5, NUMBER_OF_LEDS_PER_FAN);

  ledController.addLEDs(0, ledsChannel1, 126);

  ledController.onUpdateHook(0, []() {
      CLP::SegmentScaling segments[6] = {{21, 30}, {21, 30}, {21, 30}, {21, 30}, {21, 30}, {21, 30}};
      CLP::scaleSegments(&ledController, 0, segments, 6);
});

  // 3 fans on channel 2
  //                          FAN_PIN           CRGB array            offset             number of leds per fan
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_7, GRB>(ledsChannel2, NUMBER_OF_LEDS_PER_FAN * 0, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_8, GRB>(ledsChannel2, NUMBER_OF_LEDS_PER_FAN * 1, NUMBER_OF_LEDS_PER_FAN);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_9, GRB>(ledsChannel2, NUMBER_OF_LEDS_PER_FAN * 2, NUMBER_OF_LEDS_PER_FAN);

  ledController.addLEDs(1, ledsChannel1, 63);

  ledController.onUpdateHook(1, []() {
      CLP::SegmentScaling segments[3] = {{21, 30}, {21, 30}, {21, 30}};
      CLP::scaleSegments(&ledController, 1, segments, 3);
});
  
	temperatureController.addSensor(0, TEMP_SENSOR_PIN_1);
	temperatureController.addSensor(1, TEMP_SENSOR_PIN_2);
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
	fanController.updateFans();
}

DanFerras, 2021-11-09:
I hadn't said, but the code works, I've tested the channels individually, the problem is really about program space

Legion2, 2021-11-09:
To save  program space you can delete the hardware lightning code from the library

DanFerras, 2021-11-09:
Sorry, I couldn't find which library file the firmware code is in, could you help me find it?
Alternatively, due to my setup is a bit complex and I have some strips that I also want to control with another Arduino, I thought of controlling the halos simulating a NodePro and eliminating anything related to the LED control and temperature sensor.

Legion2, 2021-11-15:
https://github.com/Legion2/CorsairLightingProtocol/blob/dev/src/FastLEDController.cpp this file contains the hardware lighting effects

closed 2022-01-25 by github-actions[bot]