Legion2/CorsairLightingProtocol #254: Corsair LL series fan speed

jamesgohl, 2021-11-11:
I am not sure if this is a bug or something I am setting up wrong.
In the arduino code, I am trying to set up a fan with the correct RPM range, the desired result for me is that the quiet profile actually runs the fan at the lowest rpm which is 600, currently it is still audible.

No matter what values I choose for the RPM using:

`PWMFan fan1(PWM_FAN_PIN_1, 0, 2000);`

iCUE cannot control the fan speed correctly for the quiet profile; it is still too fast.  

I have tried increasing the RPM range and altering the setPower function in PMWFan.cpp:

```
void PWMFan::setPower(uint8_t percentage) {
	analogWrite(pwmPin, percentage / 2);
}

```

Any suggestions?

jamesgohl, 2021-11-11:
I seem to have achieved what I need by incresing the range by 500:

`PWMFan fan1(PWM_FAN_PIN_1, 0, 2500);`

and in PWMFan.cpp, reducing the power sent by 25 percent:

```
void PWMFan::setPower(uint8_t percentage) {
	analogWrite(pwmPin, percentage -25);
}
```
It would seem cleaner this way, quiet is low and extreme is at the fastest (audible test based on the initial fan speed when the controller starts up).

Legion2, 2021-11-12:
Maybe you can log the percentage value iCUE is using in quiet mode. If it is a high value it is a problem of iCUE.

jamesgohl, 2021-12-03:
I think more the issue was that iCUE couldn't read the fan RPM from the device, it seems to be reading the RPM now and I don't know what has changed, I've put it down to inconsistent behaviour of iCUE.  Whilst I can see the RPM now, the LL fan seems to run out of range of the specs, removing the -25 on percent and changing the range of the fan to 600 - 2500 has it working where I want it.

closed 2021-12-03 by jamesgohl