Legion2/CorsairLightingProtocol #255: Support other MCU through 16U2

cnn123666, 2021-11-14:
I designed the 16U2 as a module
[image]
I'd like to upload HoodLoader2UnoMegaController to other muc

Legion2, 2021-11-14:
The 16u2 doesn't have enough memory

cnn123666, 2021-11-14:
I want to use ATmega16U2 to handle communication, AND I want to upload sketches to ATmega644, or some other MCU

cnn123666, 2021-11-14:
[image]

Legion2, 2021-11-14:
Did you try to upload the controller sketch to the ATmega644 and have an error, or what is currently the problem?

cnn123666, 2021-11-14:
I try to compile HoodLoader2UnofficialMegaController, but the compiler error
[image]
[image]
Maybe this ATmega644 doesn't support HID Bridge?

Legion2, 2021-11-14:
The problem is that the file `HID.h` does not exist for your board. It must not contain an actual implementation, but must exist.

cnn123666, 2021-11-15:
Thanks for your reply. I understand that if my chip doesn't have HID. H, I can't send data through 16U2, right?

cnn123666, 2021-11-15:
Maybe I can try 8266, it seems to have HID. H

Legion2, 2021-11-15:
You can try to add the HID.h file next to your sketch file

Legion2, 2021-11-15:
Else I will need to update the library to exclude this file. But currently I have no time for this.

cnn123666, 2021-11-15:
Thank you very much. I already know what I'm going to do, I'm going to end this problem until I get something new, okay

closed 2021-11-15 by cnn123666