Legion2/CorsairLightingProtocol #256: device choiced - Commander Pro not working

orfin2, 2021-11-22:
**Describe the problem**
When i uploaded this sketch with choiced device Commander Pro, in my windows system devices is visible "commander pro", but iCUE app is blank without visible device. After uploading sktech blinking still orange led´s TX and RX (without running iCUE app, if i run iCUE no changes, still blinking, exiting iCUE app...blinking).
If i choiced other devices like: LS100, LT100, Lighting Node Pro and Lighting Node Core all is ok, all devices are visible in Windows devices and are also visible in iCUE app (TX and RX led´s blinking only if i run iCUE app). 

**System (please complete the following information):**
 - OS: Windows 10 Pro for Workstations, 20H2
 - Board: Arduino Leonardo
 - Device: Commander Pro
 - Version of IDE: Arduino IDE 1.8.16
 - Version of CLP: 0.14.2
 - Version of CLP Boards: 0.3.1
 - Sketch: original
 
**Code changes**
Did you changed the code of the library or the sketch? Original sketch without changes.

Legion2, 2021-11-22:
Maybe there is another application on your computer that is using the Commander PRO, so it cannot be used by iCUE?

What arduino board do you have exactly? When you switched to device types, did you also used the other sketches?

orfin2, 2021-11-23:
Yes i wrote it in system description...Arduino Leonardo. And no other software, it running on my working notebook Dell Precision 7730 (dont have integrated any leds, or RGB)...but i'll trying it on other PC with new and "clear" OS.

closed 2021-11-23 by orfin2

Legion2, 2021-11-23:
There are multiple sketches, which one did you upload?

orfin2, 2021-11-23:
if I understand correctly, i must change choice Arduino IDE in the File menu->Examples->Corsair Lighting Protocol-> from LightingNodePro to CommnderPro? Is it truth ?

Legion2, 2021-11-23:
Yes, you must select the correct sketch for the device type

orfin2, 2021-11-23:
So ok, thx for information and when i'll be home i'll try it and give feedback 👍

orfin2, 2021-11-23:
yes, you had right, now is all fully function...that was my fail, thank you for great infromation