Legion2/CorsairLightingProtocol #257: Access r, g & b colours of CRGB array (ledChannel)

jamesgohl, 2021-12-02:
**Describe the problem**
I am not a proficient c++ programmer, I work in C#.
I am using an arduino pro micro (lenorardo) cheap clone.  The device works well with this project of yours.

I have written a small function for controlling a single non-addressable RGB led (common negative).  I am attempting to access the colour values (r, g, b) of the first led from the ledsChannel array.

```
void setRGB_Button(CRGB *led) {
  if (btn_rgb_timer < millis())
  {
    uint8_t r = led.r; // breaks the program
    uint8_t g = led.g; // breaks the program
    uint8_t b = led.b; // breaks the program
    digitalWrite(BTN_R_PIN, r);
    digitalWrite(BTN_G_PIN, g);
    digitalWrite(BTN_B_PIN, b);
    btn_rgb_timer = millis() + BTN_RGB_DELAY;
  }
}
```

This is called from the loop function is as follows:

```
void loop() {
  setRGB_Button(&ledsChannel2[0]);

  cHID.update();

  if (ledController.updateLEDs()) {
    FastLED.show();
  }
}
```

The problem I have is, where ever I introduce reference to the r, g & b values of ledsChannel2[0] the program doesn't run, it bricks the  device with no COM port appearing in Windows device manager any longer. I have to short the reset and ground pins twice at power-up to bring it back to life.

What am I doing wrong? ... why can't I access the r, g & b properties of the CRGB struct, if I look at the fastLed documentation, this is the correct way to access the values.

Legion2, 2021-12-03:
Change the Datatype of the led variable to no be a pointer, remove the & from `setRGB_Button(&ledsChannel2[0])`. Also there is an example sketch in the example directory which already implements your use case.

jamesgohl, 2021-12-03:
Thank you Leon, I have it working based on your example, apologies for my ignorance I didn't think to check if there was an existing example.  

Just out of interest, why couldn't I access the CRGB colour properties directly?  Is it because the values are set in an interrupt?

Legion2, 2021-12-03:
I don't think interrupts can break it in such a way, maybe you used the wrong syntax for what you wanted to do.

jamesgohl, 2021-12-03:
What happens if you write this line in loop()…?

`analogWrite(3, ledsChannel2[0].r);`

Legion2, 2021-12-03:
this should work

closed 2022-01-25 by github-actions[bot]