Legion2/CorsairLightingProtocol #258: not recognized as LS100 even though LS100 sketch was used

Jimvanhazendonk, 2021-12-05:
**Describe the bug**
I uploaded the LS100 Sketch found in the examples to my Arduino Pro Micro but in the current version of ICUE it is recognized as a lightning node Pro which does not feature the LS100 options. I did give it a unique device ID.
Is there something wrong with the code or did I forget to enable something? I'm quite new to this and Arduino in general. I would like to use strips on the back of my screen like the LS100. 
I soldered a WS2812b strip with 21 LEDs to the Arduino to test out.

**To Reproduce**
Steps to reproduce the behavior:
upload the example for the LS100 to an Arduino pro micro, solder a 21 less WS2812 strip and connect to ICUE 4.17.224

**Expected behavior**
I uploaded the LS100 Sketch to my Arduino Pro Micro but in the current version of ICUE it is recognized as a Pro node. which does not feature the LS100 options. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: [e.g. Arduino pro micro 5v (leonardo)]
 - Device: [e.g. LS100]
 - Version of IDE: [e.g. Arduino IDE 1.8.16]
 - Version of CLP: [e.g. 0.14.2]
 - Version of CLP Boards: [e.g. 0.3.1]
 - Sketch: [LS100]

**Code changes**
NO

Legion2, 2021-12-06:
Hi @Jimvanhazendonk, you missed a step in the setup guide. You have to select the correct CLP board for the LS100 in the Arduino IDE before uploading. The it should be recognized as LS100.

Jimvanhazendonk, 2021-12-09:
Hi @Legion2,
Thank you for the tip. I did indeed forget that step.
Guess I was so excited to test it out that I rushed it :)
I have it now working with 132 LEDs. it's awesome.

closed 2021-12-09 by Legion2