Legion2/CorsairLightingProtocol #260: Is it possible to control 12 led strips individually via one sparkfun pro micro (clone)

ugene1980, 2021-12-13:
**Describe the problem**
Would like to power and control 12 LED Strips (ws2812b) via one sparkfun pro micro (clone)

Would like to ask:
1. Is this possible with one controller?
2. Do I need to connect external power to the pro micro, to drive the leds (9 RGB leds per strip)

**Screenshots**
Would wiring the LED strips in this manner be the right step to achieve it
[image]

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: sparkfun pro micro (clone)
 - Device: Lighting Node PRO

**Code changes**
n/a

Legion2, 2021-12-13:
yes, you can add multiple led strips with FastLED, independenlty of the configuration in iCUE, each real led strip will be a segment of one iCUE led strip. use `FastLED.addLeds<LEDTYPE, PIN>(leds, OFFSET, STRIP_LENGTH);` for each led strip and adjust the values, see the existing addLeds call in the examples for how to use it.

ugene1980, 2022-01-04:
> yes, you can add multiple led strips with FastLED, independenlty of the configuration in iCUE, each real led strip will be a segment of one iCUE led strip. use `FastLED.addLeds<LEDTYPE, PIN>(leds, [...]

**This works great, but I hit an unrelated snag:**
_Sketch uses 30070 bytes (104%) of program storage space. Maximum is 28672 bytes.
Global variables use 1595 bytes (62%) of dynamic memory, leaving 965 bytes for local variables. Maximum is 2560 bytes._

**Anyone can suggest anyway to reduce the size of the sketch? My sketch is currently like this:**

#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define NUM_LEDS 12

#define DATA_PIN_FAN_1 2
#define DATA_PIN_FAN_2 3
#define DATA_PIN_FAN_3 4
#define DATA_PIN_FAN_4 5
#define DATA_PIN_FAN_5 6
#define DATA_PIN_FAN_6 7
#define DATA_PIN_FAN_7 8
#define DATA_PIN_FAN_8 9
#define DATA_PIN_FAN_9 10
#define DATA_PIN_FAN_10 16

CRGB ledsChannel1[72];
CRGB ledsChannel2[48];

CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
	// 6 fans on channel 1
	//                          FAN_PIN           CRGB array     offset     number of leds per fan
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel1, NUM_LEDS * 0, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_2, GRB>(ledsChannel1, NUM_LEDS * 1, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_3, GRB>(ledsChannel1, NUM_LEDS * 2, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_4, GRB>(ledsChannel1, NUM_LEDS * 3, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_5, GRB>(ledsChannel1, NUM_LEDS * 4, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_6, GRB>(ledsChannel1, NUM_LEDS * 5, NUM_LEDS);

  // 4 fans on channel 2
  //                          FAN_PIN           CRGB array     offset     number of leds per fan
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_7, GRB>(ledsChannel2, NUM_LEDS * 0, NUM_LEDS);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_8, GRB>(ledsChannel2, NUM_LEDS * 1, NUM_LEDS);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_9, GRB>(ledsChannel2, NUM_LEDS * 2, NUM_LEDS);
  FastLED.addLeds<WS2812B, DATA_PIN_FAN_10, GRB>(ledsChannel2, NUM_LEDS * 3, NUM_LEDS);
  
	  ledController.addLEDs(0, ledsChannel1, 72);
	  ledController.addLEDs(1, ledsChannel2, 48);
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}

Legion2, 2022-01-04:
Does it make a difference if you put all strips on channel 1?

ugene1980, 2022-01-04:
> Does it make a difference if you put all strips on channel 1?

**Strangely now at 105% with this sketch(all on channel 1):**

#include <CorsairLightingProtocol.h>
#include <FastLED.h>

#define NUM_LEDS 12

#define DATA_PIN_FAN_1 2
#define DATA_PIN_FAN_2 3
#define DATA_PIN_FAN_3 4
#define DATA_PIN_FAN_4 5
#define DATA_PIN_FAN_5 6
#define DATA_PIN_FAN_6 7
#define DATA_PIN_FAN_7 8
#define DATA_PIN_FAN_8 9
#define DATA_PIN_FAN_9 10
#define DATA_PIN_FAN_10 16

CRGB ledsChannel1[120];

CorsairLightingFirmware firmware = corsairLightingNodePROFirmware();
FastLEDController ledController(true);
CorsairLightingProtocolController cLP(&ledController, &firmware);
CorsairLightingProtocolHID cHID(&cLP);

void setup() {
	// 6+4 fans on channel 1
	//                          FAN_PIN           CRGB array     offset     number of leds per fan
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_1, GRB>(ledsChannel1, NUM_LEDS * 0, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_2, GRB>(ledsChannel1, NUM_LEDS * 1, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_3, GRB>(ledsChannel1, NUM_LEDS * 2, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_4, GRB>(ledsChannel1, NUM_LEDS * 3, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_5, GRB>(ledsChannel1, NUM_LEDS * 4, NUM_LEDS);
	FastLED.addLeds<WS2812B, DATA_PIN_FAN_6, GRB>(ledsChannel1, NUM_LEDS * 5, NUM_LEDS);
        FastLED.addLeds<WS2812B, DATA_PIN_FAN_7, GRB>(ledsChannel1, NUM_LEDS * 6, NUM_LEDS);
        FastLED.addLeds<WS2812B, DATA_PIN_FAN_8, GRB>(ledsChannel1, NUM_LEDS * 7, NUM_LEDS);
        FastLED.addLeds<WS2812B, DATA_PIN_FAN_9, GRB>(ledsChannel1, NUM_LEDS * 8, NUM_LEDS);
        FastLED.addLeds<WS2812B, DATA_PIN_FAN_10, GRB>(ledsChannel1, NUM_LEDS * 9, NUM_LEDS);
	
  ledController.addLEDs(0, ledsChannel1, 120);
}

void loop() {
	cHID.update();

	if (ledController.updateLEDs()) {
		FastLED.show();
	}
}

closed 2022-01-26 by github-actions[bot]