Legion2/CorsairLightingProtocol #261: Displaying artifacts (rainbow) for Node Pro

konradbjk, 2021-12-26:
**Describe the problem**
While setting transitions for colours in iCUE for a second there is a rainbow on LEDs displayed - kind of artifacts. Like the lights were clearing itself. 

I have seen past issues and there was similar case for LS100. It was due to brightness. I tried modifying brightness in device settings. There are only 3 levels of brightness, without % indication. Though, updating brightness in iCUE make effect on LEDs brightness.

**System (please complete the following information):**
 - OS: Windows 10
 - Board: Arduino Leonardo ETH pins 2&3
 - Device: Lighting Node PRO
 - Version of IDE: latest Arduino IDE for macOS
 - Version of CLP: latest
 - Version of CLP Boards: latest
 - Sketch: Node Pro
- iCUE: the latest for windows.

Legion2, 2021-12-26:
What do you mean by "transitions for colours", which effect in iCUE do you use. Does it happen for hardware lighting? Does it only happen while you  change the settings in iCUE or all the time?

konradbjk, 2021-12-31:
It seems to not matter what effect it is. Even if I set static color, then change the color, the rainbow artifacts happen. It happens for both software and hardware lightning.

What might be helpful is that those artifacts happens (sometimes) only for some of the LEDs, other just go dark in that time.

Legion2, 2021-12-31:
The rainbow artifacts you see are probably the default effects of the hardware lighting mode, which is activated for a short time while you change the settings.

konradbjk, 2021-12-31:
I thought so, then I disabled the hardware effects and had only software ones. Still the same. 

Is there a way to debug the signal/data?

Legion2, 2021-12-31:
You can not disable the default hardware lighting, it is the default effect which is shown if nothing else is configured. 
To debug the Arduino, you can enable the debug mode by setting the DEBUG parameter during compilation. There are also other parameter which can be enabled to show more info on the serial output. Have a look at the debug sketch.

Legion2, 2021-12-31:
https://github.com/Legion2/CorsairLightingProtocol/blob/dev/examples/DebugSketch/board.txt here are the flags which can be set. To enable them, you have to edit the board.txt file which is in the folder were you installed the CLP boards.

closed 2022-01-25 by github-actions[bot]