Legion2/CorsairLightingProtocol #263: Cannot change emulated device type

netdudeuk, 2022-01-08:
Hi

I created a simulated Lighting Node Pro with eight LEDs and that works ok thanks.

I decide to download the LT100 sketch (changing the number of LEDs to eight) but iCUE says that I still have a LNP after the Done uploading message.

Any ideas please ?

Thanks

Legion2, 2022-01-08:
You have the select the correct board as shown here https://github.com/Legion2/CorsairLightingProtocol/blob/dev/extra/images/select-board.png

netdudeuk, 2022-01-08:
Thanks for the prompt reply.

I checked the setting and discovered that there's a Device setting that also has to be changed.  So, my mistake.  I do now see the device in iCUE but the LED Detection Run button doesn't detect the 'tower'.  I must be missing something else.

Legion2, 2022-01-08:
What 'tower' do you have? The detection feature is implemented in such a way that it uses the number of configured leds in the sketch

closed 2022-01-30 by github-actions[bot]