Legion2/CorsairLightingProtocol #266: Clarification for wiring & Control

Jonathan-Garber, 2022-01-27:
### Goal
Trying to be sure I understand the wiring/control options here with my hardware and need clarification on a few things.
I am considering purchasing a new computer case with 6 aRGB Fans and built in control board.

I want to use iCUE and have individual control and sync control via iCUE software.

### Hardware

_SparkFun Pro Micro 5v/16MHz -- Setup as a Lightning Node Pro_
[image]

_6 aRGB 5v 4pin connector LED strips***_ 
***Note: They are actually 5v aRGB fans with 3 physical pins on a 4 pin connector. See image below for pinout
[image]

**Optional Hardware**
Included control board for the aRGB Leds
[image]

**Questions**
This cases included controller board. I can see the Pins for connecting to the motherboard are labeled I need clarification of a few.

***NOTE: The power is connected above this connector in a 12v | GND | 5v pinout. That one is clearly understood and powers aRGB and Fans connectors

[image]

I assume the following from top down.

GND == Ground - from motherboard header ( aRGB Leds )
D-IN == Data IN - from motherboard header ( aRGB Leds )
GND == Ground but from where?  possible ground from motherboard PWM header ( PWM Fan connectors )
P-IN == unknown | possible RPM from PWM Fan to board  ( PWM Fan connectors )
PC -IN == unknown | possible PWM from motherboard PWM header to Fan ( PWM Fan connectors )

This is my assumption for the above pinout. I don't have the physical board here yet to investigate but if anyone has seen or used a similar board is this correct? Also if you know, is this board actually wired in series to the aRGBs or just parallel?

### Wiring

_How do these need to be wired and connected to the controller board LNP ( Sparkfun Pro Micro )_

My understanding is these fans are individually controllable as long as they are daisy chained together in series, NOT parallel.
As the control command has to pass from strip 1 D Pin to the next strips D pin and so on until end of the chain.

```
LNP Pin 2 --> LED 1 Pin D
LED 1 Pin D --> LED 2 Pin D
LED 2 Pin D --> LED 3 Pin D
LED 3 Pin D --> LED 4 Pin D
```
Repeat this wiring in series until we hit the end of the chain.

If the optional control board included with this new case has these plugs wired in series I should be able to tack into its D-In Pin from the LNP board.

**Ideal Wiring**
```
LNP Pin 2 --> Control Board Pin D-IN
```
This would in theory go down the connectors D Pins on the board in series ( I don't have the physical board here yet to check pins/traces )

Is my understanding of the wiring/control scheme above correct?
Any input / comments are appreciated. Thanks guys.

Legion2, 2022-01-27:
Where is the other side of the optional colleges controll board with the unknown pins plugged in? Does it connect to the motherboard RGB header or USB?

Your wiring is correct, if the optional board is not included and the led fans do not support daisy chaining, you can connect each individually to the Arduino, there is an example sketch for that.

Jonathan-Garber, 2022-01-27:
> Where is the other side of the optional colleges controll board with the unknown pins plugged in? Does it connect to the motherboard RGB header or USB?
>
> Your wiring is correct, if the optional board [...]

I don't have any of the hardware here yet. Just planning ahead of time. So I haven't seen the types of plugs on the other end yet or the fans plugs.

It just now clicked in my head that each fan has to have 3 plugs to daisy chain correctly.
1. Fan connector
2.  aRGB IN
3. aRGB Out ( to daisy chain )

I don't think the manufacturer provided fans are daisy chainable - that's why they included the 6 port aRGB control board. So I would have to wire each fans D pin to its own pin on the LNP board. That's kind of annoying.

closed 2022-01-28 by Jonathan-Garber