Legion2/CorsairLightingProtocol #271: How to convert LC100 to normal strip format

cnn123666, 2022-02-17:
[image]

cnn123666, 2022-02-17:
https://user-images.githubusercontent.com/52467395/154504312-42360867-565c-48e2-bd9d-bea86a6d93c8.mp4

This is his order. 4,5,6,7,8,9,1,2,3
[image]
I notice this function, but I don't know how to change it

Spegs21, 2022-02-18:
It'll be a very similar function. You'll just have to std::rotate the array in groups of 9 enough times to put it in order. I can probably write it if I get a chance.

cnn123666, 2022-02-19:
LIGHTING NODE PRO now supports 18 LC100 (162) and 6 QL fans (204). A maximum of 366 beads

Spegs21, 2022-02-19:
@cnn123666 Are you sure? If I select 6 QL fans on one channel it disables the other channel with the Lighting Node Pro sketch.

cnn123666, 2022-02-19:
You can add 18 LC100s in channel 1, and you can add other devices in channel 2.

cnn123666, 2022-02-19:
Can the

cnn123666, 2022-02-19:
[image]

Spegs21, 2022-02-19:
Interesting... Is this with a newer update to iCue? I'll have to look into it.

cnn123666, 2022-02-20:
@Spegs21 Update to the latest ICUE

Spegs21, 2022-02-24:
@cnn123666 I'm convinced this is a bug with iCue channel set up and LC100. In your last screenshot, if you were to click the dropdown for channel 1, the only option it would give you is "No Device Connected". The LED limiting channel interactions are not working properly like they do for other devices. If you changed your channel 2 to 5 QL fans and changed your device on channel 1 to something else then back to LC100 it will only let you add 2 companion triangles.

cnn123666, 2022-03-01:
I don't think it's a BUG 。Now the ICUE UI displays normally
[image]

Spegs21, 2022-03-01:
This has less total LEDs (297) than the setup with 6 QL fans and 18 LC100s so it's probably allowed.

Spegs21, 2022-03-03:
@cnn123666 Can you test the function from my repo? CLP::TransformLC100ToStrip

cnn123666, 2022-03-04:
Can I only use it on the 32U4？
[image]

cnn123666, 2022-03-04:
@Spegs21 Hello! I tested it on the 32U4 and it worked. Also fixed to light strip format. But it doesn't compile on pico

Spegs21, 2022-03-04:
@cnn123666 It compiled OK on pico for me. It looks like your build log is showing it's trying to include a wifi library. It shouldn't be doing that. Do you have an extra include at the top?

cnn123666, 2022-03-04:
@Spegs21 I'm sorry for my problem. After the Arduino was restarted, the compilation was successful. Thank you very much for your help.

closed 2022-03-04 by cnn123666