Legion2/CorsairLightingProtocol #280: 'CorsairLightingFirmwareStorageEEPROM' does not name a type; did you mean 'CorsairLightingFirmwareStorageStatic'?

skylinegodzilla, 2022-03-13:
**Describe the problem**
I try to build the LightingNodePro Code however I get the error 
'CorsairLightingFirmwareStorageEEPROM' does not name a type; did you mean 'CorsairLightingFirmwareStorageStatic'?
Every time I try to compile.
However, the .h clearly exists in the Arduino > libraries > Corsair_lighting_protocol directory

I have instlled the CLP bords.

On a side note, I have unos, NodeMCUs, IOT 33's, and lilypads, I wish there was a way to get tiny USB to work with the IOT 33 but sadly I can't figure out how to so I just guess ill have to wait for Arduino to add support for tiny USB to there core. I don't want to risk flashing one of my Unos until I can at least get the code to compile. And I have no idea if It will work on the Lilypad, it's an 23u4 AVR board.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: Windows 11
 - Board: Explained above
 - Device: Lighting Node PRO/smart lighting controler
 - Version of IDE: Arduino IDE 1.8.16
 - Version of CLP: 0.15.0
 - Version of CLP Boards: ??? lattest?
 - Sketch: LightingNodePRO

**Code changes**
None

**Full error report*
Arduino: 1.8.16 (Windows 10), Board: "CLP Arduino Zero (Native USB Port), Lighting Node PRO"

In file included from C:\Users\benca\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src/FastLEDController.h:18:0,

                 from C:\Users\benca\Documents\Arduino\libraries\Corsair_Lighting_Protocol\src/CorsairLightingProtocol.h:34,

                 from C:\Users\benca\AppData\Local\Temp\arduino_modified_sketch_288582\LightingNodePRO.ino:16:

C:\Users\benca\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

LightingNodePRO:25:1: error: 'CorsairLightingFirmwareStorageEEPROM' does not name a type; did you mean 'CorsairLightingFirmwareStorageStatic'?

 CorsairLightingFirmwareStorageEEPROM firmwareStorage;

 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

 CorsairLightingFirmwareStorageStatic

LightingNodePRO:26:62: error: 'firmwareStorage' was not declared in this scope

 CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);

                                                              ^~~~~~~~~~~~~~~

C:\Users\benca\AppData\Local\Temp\arduino_modified_sketch_288582\LightingNodePRO.ino:26:62: note: suggested alternative: 'firmware'

 CorsairLightingFirmware firmware(CORSAIR_LIGHTING_NODE_PRO, &firmwareStorage);

                                                              ^~~~~~~~~~~~~~~

                                                              firmware

LightingNodePRO:27:1: error: 'FastLEDControllerStorageEEPROM' does not name a type; did you mean 'FastLEDControllerStorage'?

 FastLEDControllerStorageEEPROM storage;

 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

 FastLEDControllerStorage

LightingNodePRO:28:34: error: 'storage' was not declared in this scope

 FastLEDController ledController(&storage);

                                  ^~~~~~~

LightingNodePRO:30:1: error: 'CorsairLightingProtocolHID' does not name a type; did you mean 'CorsairLightingProtocolSerial'?

 CorsairLightingProtocolHID cHID(&cLP);

 ^~~~~~~~~~~~~~~~~~~~~~~~~~

 CorsairLightingProtocolSerial

C:\Users\benca\AppData\Local\Temp\arduino_modified_sketch_288582\LightingNodePRO.ino: In function 'void loop()':

LightingNodePRO:40:2: error: 'cHID' was not declared in this scope

  cHID.update();

  ^~~~

C:\Users\benca\AppData\Local\Temp\arduino_modified_sketch_288582\LightingNodePRO.ino:40:2: note: suggested alternative: 'HI2'

  cHID.update();

  ^~~~

  HI2

exit status 1

'CorsairLightingFirmwareStorageEEPROM' does not name a type; did you mean 'CorsairLightingFirmwareStorageStatic'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Legion2, 2022-03-13:
If you want to use Arduino boards, look in the wiki, there is a special guide for that.

skylinegodzilla, 2022-03-14:
Ive looked into the wiki it only covers unos and mega both of witch require flashing witch I said I don't want to risk doing until I can get the original code to compile first.

Not onley this but i have also dug deeper then just the wiki ive looked in to your pr's and have found where you added support for sand boards by utilising tinyUSB however ardrino sandbords dont support tinyUSB in there core so it onley helps 3rd party sand bords like the adafruit fether.

There is nothing in the wiki that explains how to get ardrino sand bords to work. And i am willing to flash one of my unos but first i need to make sure that i can at least get the existing code to compile. 

So again how do i solve the issue 

CorsairLightingFirmwareStorageEEPROM' does not name a type; did you mean 'CorsairLightingFirmwareStorageStatic'?

Spegs21, 2022-03-14:
@Legion2 He is trying to use "CLP Arduino Zero (Native USB Port)". It won't work because support for SAMD is implemented through TinyUSB. I thought I had a PR in to remove the SAMD boards from CLP boards but I guess I didn't. I can create a PR for this.

@skylinegodzilla CLP boards (SAMD) won't work. It's leftover from an attempt at a different implementation. SAMD boards, like the Zero, are only supported through TinyUSB and don't have an eeprom. You'd have to use the static storage class like in the TinyUSB example. 

You should be able to use TinyUSB on this board by installing the adafruit library per the Readme and manually defining USE_TINYUSB in the build parameters. Take a look a the way Legion did this with the boards.txt file in his debug sketch example.

Legion2, 2022-03-14:
@Spegs21 yes, please create the PR and thanks for your help

skylinegodzilla, 2022-03-14:
@Spegs21 
You said you could leave an example on how to set up tinyUSB for unsupported cores? Please do I would love it if I could use one of my ardrino IOT 33 SAMD boards but i could not figure out the core part for tinyUSB

Spegs21, 2022-03-15:
@skylinegodzilla It shouldn't be a problem. I should be able to get to it later this week.

skylinegodzilla, 2022-03-15:
@Spegs21  Thank you so much.

skylinegodzilla, 2022-03-20:
@Spegs21 Hello is There an update on that example?

Spegs21, 2022-03-20:
@skylinegodzilla I started on this but then realized it was going to be harder than I expected. TinyUSB conflicts with the Arduino SAMD core USB implementation and there isn't a way to disable it currently. I started arduino/ArduinoCore-samd#666 to see if they'd be open to at least allow the possibility to use it. It can still be done but it's going to take more time than I expected.

skylinegodzilla, 2022-03-21:
@Spegs21 ok thank you and im guessing there's no way to get this to work using the arduino samd core?

skylinegodzilla, 2022-03-22:
@Spegs21  wow you posted that request to them 5 days ago and they still have not replayed. I would expect them to be more proactive then that.

I'm very appreciative that you guys are responsive.

Spegs21, 2022-03-22:
@skylinegodzilla These things can take some time. I think it's maintained mostly by volunteers.

I did make some changes to the core as I described for testing but there are conflicting definitions of Serial between the core and TinyUSB. So one or the other would need changed and the core definition of Serial is actually in the Arduino API, a different repo.

Spegs21, 2022-03-23:
It looks like the TinyUSB Arduino library already has logic to define the CDC serial differently in some cases. It will just need updated to trigger the change when using the Arduino SAMD core.

closed 2022-04-14 by github-actions[bot]