Legion2/CorsairLightingProtocol #281: Raspberry 2040 and TinyUSB

MakerVejby, 2022-03-23:
After uploading the sketch to the RP2040, the RP2040 presents itself as a USB to Serial (TinyUSB Serial).

If I start **USBview** I can see the following.. 
_Device Descriptor:
bcdUSB:             0x0200
bDeviceClass:         0xEF
bDeviceSubClass:      0x02
bDeviceProtocol:      0x01
bMaxPacketSize0:      0x40 (64)
idVendor:           0x1B1C
idProduct:          0x1C1A
bcdDevice:          0x0100
iManufacturer:        0x01
iProduct:             0x02
0x0409: "Lighting Node PRO"
iSerialNumber:        0x03
0x0409: "FB66DF55421900F5"
bNumConfigurations:   0x01_

But i can also see in **usbview**
_Interface Descriptor:
bInterfaceNumber:     0x00
bAlternateSetting:    0x00
bNumEndpoints:        0x01
bInterfaceClass:      0x02
bInterfaceSubClass:   0x02
bInterfaceProtocol:   0x00
iInterface:           0x04
0x0409: "TinyUSB Serial"_

**System (please complete the following information):**
 - OS: [e.g. Windows 7 and10]
 - Board: [e.g. RP2040]
 - Device: [e.g. TinyUSB (Corsair Lightning Node pro)]
 - Version of IDE: [e.g. Arduino IDE 1.8.19]
 - Version of CLP: [e.g. 0.15.0]
 - Version of CLP Boards: [e.g. 0.3.1]
 - Sketch: [e.g. TinyUSB]

I have added the 6 files for the RP2040 and #elif defined(ARDUINO_ARCH_RP2040) / #include statements in [led_sysdefs.h] and [platforms.h]

Legion2, 2022-03-23:
@Spegs21 can you help here?

Spegs21, 2022-03-23:
The CDC endpoint is not disabled in the code and it will be visible along side the HID endpoint. It should not affect the operation of the device. It is needed to be able to upload to the Pico without manually entering the bootloader. 

Is the device not visible in iCue?

MakerVejby, 2022-03-24:
No, The device is not visible in ICue / device manager.
Only "TinyUSB Serial" is visible in device manager.

When i first programmed the rp2040 and when the sketch was uploaded i could see "Lightning node pro" as a device. But when windows has searched for driver only "TinyUSB Serial" is displayed.

Spegs21, 2022-03-24:
@MakerVejby Have you me any changes to the sketch? Have you used this board or a different board as a different Corsair device through this library on your PC? Sometimes iCue won't show the device without a restart because the device ID it remembers is for a different device.

MakerVejby, 2022-03-24:
I have only used this board for this program. I have not made any changes in the code.
I have programmed one Arduino Pro Micro and that one works like a charm.

I have tried to reboot the computer but it didn't hel.

MakerVejby, 2022-03-24:
Of course not! :-) That would be a easy fix.

MakerVejby, 2022-03-24:
> The CDC endpoint is not disabled in the code and it will be visible along side the HID endpoint. It should not affect the operation of the device. It is needed to be able to upload to the Pico without manually entering the bootloader.
> 
> Is the device not visible in iCue?

Spegs21, 2022-03-24:
@MakerVejby It doesn't look like the support for entering the bootloader with the serial trigger is implemented in the Pico core. Adafruit has this working for their SAMD boards though. 

I compiled and uploaded the TinyUSB sketch to my Pico through the Arduino IDE and it is working fine. Can you enable verbose output during compilation in the IDE preferences and post the build log here?

MakerVejby, 2022-03-25:
Please see txt file
[Compile.txt](https://github.com/Legion2/CorsairLightingProtocol/files/8348831/Compile.txt)

MakerVejby, 2022-03-25:
When i try to change serial according to the example i only get "'**CorsairLightingProtocolHID' does not name a type; did you mean 'CorsairLightingProtocolSerial'?**"

Spegs21, 2022-03-25:
@MakerVejby You need to pass the serial number to the CorsairLightingProtocolTinyUSBHID constructor not add the CorsairLightingProtocolHID constructor. I just said you do it in the same way.

MakerVejby, 2022-03-25:
@Spegs21 Sorry, My bad. I was to eager to get it to work on the RP2040.
I have changed the code so it's correct now. But no success.
I've tried the RP2040 on two other computer that have newly installed windows 10 and 7 and the latest Icue.
But still the same. It won't show up in Icue as a device. I have tried with new serial and new device id.

Spegs21, 2022-03-25:
@MakerVejby can you enable debug logging in iCue and export the logs after plugging in the device then post the zip file here?

Spegs21, 2022-03-25:
@MakerVejby I just noticed your idProduct is incorrect.

idProduct: 0x1C1A

Did you change some of the files? This should be 0x0C0B

MakerVejby, 2022-03-25:
Well, look at that! Now it's working! Show's up as a Corsair Node Pro! Don't remember changing the PID but when i look at the files on github they differs from my files. So i must say that the file has been altered here. 

Thank you for all the support!

closed 2022-03-25 by MakerVejby