Legion2/CorsairLightingProtocol #283: Is it possible to connect any argb device with the arduino or do these work different?

AquaCobell, 2022-03-26:
I was thinking of linking all my argb devices including some fans and the lian li strimmer plus cause msi mystic light is really bad, is this theoretically possible or do these work different than normal ws2812 stripes?

Legion2, 2022-03-26:
yes theoretically its possible to link any device, as long as you know the type of leds it uses and how its wired.

AquaCobell, 2022-03-26:
thanks that sounds very good, what do you exactly mean by types of leds? I thought every argb devices uses ws2812 or something working the same, or could they have different ones when using an argb header?

Legion2, 2022-03-27:
Yes they use some kind of ledchipset, which can be configured with the FastLED library which is included in every example sketch.

JoseArtunedo, 2022-04-09:
In principle this would be to emulate iCue devices and studio.

If you have the original devices, there are other projects; where it identifies all the devices and synchronizes them all, without having to use MystiLight, iCue, etc.

closed 2022-05-01 by github-actions[bot]