Legion2/CorsairLightingProtocol #284: Is it possible to have more than 2 LED channels, or is it possible to run “two instances” from a single Arduino?

Inkedxx, 2022-04-04:
Not sure if this is the correct place to post this but I’m new to GitHub so I apologize. If not the correct place for this, I’d appreciate some direction. Thank you.

I am brand new to RGB lighting, controllers and hardware programming. I recently built my first PC and decided to go with the Corsair lighting platform. My RGB capable devices and their connections are:

• ASUS TUF Gaming B660 Plus Wifi D4 (Aura Sync)
• 2x 1TB XPG NVMe SSD ~> RGB control from Mobo (Aura Sync)
• 24 pin ATX 90° adapter ~> Connected to Mobo ARGB Header (Aura Sync)
• Lian Li Strimer (24 & 6 pin) ~> Lian Li Controller ~> Corsair Commander Pro (iCUE)
• Lian Li Galahad 240 AIO ~> Corsair Commander Pro (iCUE)

I would like to migrate all RGB devices not currently controller by iCUE to use iCUE. I also want to have all future RGB devices to also be controlled by iCUE. To accomplish this, I need additional LED channels due there are only two available on the Commander Pro.

I saw the the Lighting Node Pro is not sold alone, but as an included item on Corsair LED strips and the LS100 triangles. The strips are low density and I don’t like how few LEDs they have and I’m not interested in the triangles…not for $150, specially when all I really want is the Lighting Node Pro.

This has led me directly to this page and the possibility of creating my own with Arduino. I have purchased the items I need for this, but while doing this, I was wondering if I should buy TWO Arduino UNO’s so I can make two of these and have 4 channels.

However, I was wondering is it possible to program the Arduino for 4 channels instead of just two?

OR, is it possible to run “two instances” of this in order to achieve 4 channels total?

Forgive my ignorance if this is simply not possible…again, this is my first time learning about all this.

Legion2, 2022-04-04:
You can only have two channels per Arduino. Also it's much easier to use Arduino Leonardo or SparkFun Pro Micro boards than Arduino uno boards.

Inkedxx, 2022-04-04:
Hi! Thank you for the reply. Can I follow the same process if I go with your suggestion of using Arduino Leonard instead of the Uno?

Legion2, 2022-04-05:
The process of setting up Arduino Leonardo is simpler than the Arduino uno. It is described here https://github.com/Legion2/CorsairLightingProtocol#create-a-lighting-node-pro-with-avr

closed 2022-04-05 by Legion2