Legion2/CorsairLightingProtocol #288: Intermittent Hyperflash

RivalSon, 2022-05-07:
Picked up a Pro Micro, got it all set up and loaded with CLP, FastLED etc.. it shows in iCUE, works to display what iCUE shows but it will intermittently just flash random colours at full brightness all the way along the strip. Using WS2812B LEDs.. 
 https://www.dropbox.com/s/8x602uxfv2h3j44/20220507_043621.mp4?dl=0

Legion2, 2022-05-07:
Looks like the strip is either not correctly wired, maybe ground is not connected to power source and arduino, or one led of the strip is broken and causes the effects down the line

RivalSon, 2022-05-07:
That might be it! I had the strip wired up for another microcontroller so just left the 5V connected to the strip but ran the signal only from the Arduino, powering it by USB. I'll pop the 5V & GND into the Arduino too and see what it does.

Legion2, 2022-05-07:
Only the GND should be connected to both Arduino and power supply. The 5v must only be connected to the power supply

RivalSon, 2022-05-07:
Just tested and working perfectly now!  Can't believe I didn't think of that! I know for future though! I think the Pro Micro can take 5V DC in (I have a 5V regulated PSU for powering long strips for testing and eventually implementation).

RivalSon, 2022-05-07:
> Just tested and working perfectly now!  Can't believe I didn't think of that! I know for future though! I think the Pro Micro can take 5V DC in (I have a 5V regulated PSU for powering long strips for testing and eventually implementation).

Damn, *almost perfectly. Still some random full brightness flashes. Only GND and Signal connected to Arduino..

closed 2022-05-22 by RivalSon