Legion2/CorsairLightingProtocol #289: Raspberry Pi Pico with TinyUSB ,An error occurred while creating more than eight ports

773534601, 2022-05-12:
**Describe the problem**
When Arduino creates more than eight ports, the first eight correctly output signals. Starting from the ninth, the light effect is mixed with error signals, resulting in flashing lights.But he can still show the right light. 
Is this a PIO timing problem?
How to avoid and solve this problem?

**Screenshots**
void setup() {
	cHID.setup();
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_1, GRB>(ledsChannel1, LEDSA * 0, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_2, GRB>(ledsChannel1, LEDSA * 1, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_3, GRB>(ledsChannel1, LEDSA * 2, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_4, GRB>(ledsChannel1, LEDSA * 3, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_5, GRB>(ledsChannel1, LEDSA * 4, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_6, GRB>(ledsChannel1, LEDSA * 5, LEDSA);

  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_7, GRB>(ledsChannel1, LEDSA * 6, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_8, GRB>(ledsChannel1, LEDSA * 7, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_9, GRB>(ledsChannel1, LEDSA * 8, LEDSA);
//  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_10, GRB>(ledsChannel1, LEDSA * 9, LEDSA);
//  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_11, GRB>(ledsChannel1, LEDSA * 10, LEDSA);
//  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_12, GRB>(ledsChannel1, LEDSA * 11, LEDSA);

  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_13, GRB>(ledsChannel2, LEDSA * 0, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_14, GRB>(ledsChannel2, LEDSA * 1, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_15, GRB>(ledsChannel2, LEDSA * 2, LEDSA);
  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_16, GRB>(ledsChannel2, LEDSA * 3, LEDSA);
//  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_17, GRB>(ledsChannel2, LEDSA * 4, LEDSA);
//  FastLED.addLeds<WS2812B, DATA_PIN_CHANNEL_18, GRB>(ledsChannel2, LEDSA * 5, LEDSA);
   
  ledController.addLEDs(0, ledsChannel1, LEDXA);  
  ledController.addLEDs(1, ledsChannel2, LEDXA); 
}
void loop() {
	cHID.update();
if (ledController.updateLEDs() ) { 
   FastLED.show(); 
    }
        
//   delay (1);
}

**System (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Board: [pico]
 - Device: [Lighting Node PRO]
 - Version of IDE: [Arduino IDE 1.8.15]
 - Version of CLP: [0.15.0]
 - Version of CLP Boards: [ ]
 - Sketch: [TinyUSB]

**Code changes**
DATA_PIN_CHANNEL_9 10 11 12....And 13 14There will be uncontrollable flashing error lights, especially when the black light is intermittent
A level shifter or buffer,  to the 5v logic level of the LEDs.
This still flashes with errors.
![U
[image]
ploading 1652333684607.gif…]()

773534601, 2022-05-12:
[image]
[image]
[image]
[image]

cnn123666, 2022-05-12:
https://github.com/Legion2/CorsairLightingProtocol/issues/273

cnn123666, 2022-05-12:
https://github.com/Legion2/CorsairLightingProtocol/issues/273#issuecomment-1100501765

closed 2022-05-27 by 773534601