Legion2/CorsairLightingProtocol #292: Support for new LC100

GodlyOverlord, 2022-06-21:
Hiya

Has there been any thought on adding direct support for the new LC100 light panels? (the nanoleaf style things)

I'm not 100% if there's anything new/fancy with effects (I assume there would be) over just using a standard LED strips though.

Thanks for making such a great project!

Legion2, 2022-06-27:
To support new devices we need some technical details about them, USB IDs and capabilities. It's best if someone has the device to provide this information. However, new icue devices work differently, see #211 and are currently not supported.

VikasMulchandani, 2022-07-12:
@Legion2 The LC100 is an amazing alternative to Nanoleaf. If you can help emulate this it will be very helpful for DIY'rs.
I have Nano Leaf emulation working on ESP with another software but it does not sync with the desktop LED. 
I just wish I could sync everything with SignalRGB.

Please do consider it, TIA

Legion2, 2022-07-12:
Will look into it when I find some time

VikasMulchandani, 2022-07-12:
Thanks much.... Dropping a coffee tip on Paypal for considering it.

Legion2, 2022-07-16:
@GodlyOverlord @VikasMulchandani You need to update iCUE 4 to the latest version (check in the iCUE settings that you have version `4.25.155` or higher) and create a LNP with this library. Then you should be able to select "LC100 Accent Triangles" in the the LNP Lighting Setup in iCUE:
[image]

GodlyOverlord, 2022-08-27:
Thats awesome! I didn't realize they were still driven via a lighting node pro, i figured they were some sort of stand alone controller like the LT's and such. 

Thanks for letting us know! Now onto 3dprinting up something that works!

closed 2022-08-27 by Legion2